commaai/openpilot #22417: 0.8.9 drifting into right lane

ynwng, 2021-10-03:
**Describe the bug**

Openpilot drifted into the right lane a few times on one of my routes.

**How to reproduce or log data**

Drive at night using the leftmost lane.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: f28c43ff6a3dea6f
 - Route: f28c43ff6a3dea6f|2021-10-02--19-48-56
 - Timestamp: 19:49:06, 19:49:25
 - Version: I already updated my device, so I'm not sure what the commit hash was. It was Shane's fork 0.8.9 though.
 - Car make/model: Toyota Corolla Hatchback 2020

**Additional context**

Maybe not important, but when openpilot was enabled at 19:49:56, it was driving right on the lane's right edge which felt uncomfortable, but at least it didn't drift into the next lane.

pd0wm, 2021-10-04:
Can you make sure the raw logs are uploaded? Turn on your device, go to https://useradmin.comma.ai/?onebox=f28c43ff6a3dea6f%7C2021-10-02--19-48-56 and click on "upload all files".

HaraldSchafer, 2021-10-06:
@nuwandavek This is another locationd biased yaw rate causing angle offset failure. Can you figure out what is causing these and come up with a solution? This could mean not using locationd in params in certain conditions or ignoring GPS if it's bad.

Mostly though can we get some metrics on how often the gyro is biased like this?