commaai/openpilot #22423: Boardd: libusb initialization error after too many connect retries

pd0wm, 2021-10-04:
If you leave boardd running with no pandas connected it will start printing errors after a few minutes and no longer connects to a panda.

```
```
libusb: error [usbi_pipe] failed to create pipe (24)
selfdrive/boardd/panda.cc: libusb initialization error
libusb: error [usbi_pipe] failed to create pipe (24)
selfdrive/boardd/panda.cc: libusb initialization error
libusb: error [usbi_pipe] failed to create pipe (24)
selfdrive/boardd/panda.cc: libusb initialization error
...
```

pd0wm, 2021-10-05:
@robbederks Mentioned creating and destroying a context leaks an fd. We have to fix that or re-use the same context

robbederks, 2021-10-05:
Indeed. Initializing and (re)using the default libusb context (pass in nullptr) fixes this

closed 2021-10-07 by pd0wm