commaai/openpilot #22429: Replay feature requests

pd0wm, 2021-10-04:
- [ ] Automatically add services to the blocklist. E.g. after hitting a multiple publishers error, don't try to broadcast anymore

deanlee, 2021-10-06:
> Ask confirmation before falling back to qlogs/qcameras, --qlog to skip confirmation

What to do after user chooses No? skip to the next non-qcamera/qlog segment or just exit?

pd0wm, 2021-10-06:
Just exiting is fine

deanlee, 2021-10-09:
@gregjhogan :  #22503

adeebshihadeh, 2021-10-15:
@deanlee seems qcams are broken, at least for C3 routes
[image]

deanlee, 2021-10-15:
It may be a transform problem (not very sure yet). I checked the image output by frameread,  it doesn't seem to have any problems.