commaai/openpilot #22500: framereader/logreader enhancements

gregjhogan, 2021-10-09:
- [ ] logreader support for processing a single segment without writing out any files
- [ ] framereader support for processing a single segment without writing out any files
- [ ] framereader support for [index video stream](https://github.com/commaai/openpilot/blob/537dff3ae0bac7ca3d40f1441297cc146dc1a9bb/tools/lib/framereader.py#L132-L144) without writing out any files

support from command line is not needed, this is for spinning through thousands of random segments as fast as possible without disk I/O becoming a bottleneck (or wearing out SSD drives)

adeebshihadeh, 2021-10-10:
@deanlee after these are done, we should be able to replace the python logreader and framereader with a wrapper around the new ones

deanlee, 2021-10-12:
I need to find a way to pass DynamicStruct to the pycapnp PyObject

closed 2023-10-07 by gregjhogan