commaai/openpilot #22562: Fix up macOS support

adeebshihadeh, 2021-10-14:
Now that WSL2 is known to work well for Windows users, we just need to fix up mac support.

- [ ] fix x86 build
- [ ] fix M1 build
- [ ] fix setup script
- [ ] get CI up again
- [ ] test replay
- [ ] build + test plotjuggler (https://github.com/commaai/PlotJuggler/issues/13)

bravech, 2021-10-15:
I have a build on M1 working in https://github.com/bravech/openpilot/commit/c037ac9a35108b161d48f97e8421dfd415ccaeca , but there are several issues with running the tools after building, such as: `ui` slows down after running for a few minutes and eventually crashes, `watch3` crashes with Qt-related errors, . One possible culprit may be that `clang` uses `libc++` instead of `libstdc++`, but I'm not sure if it would cause these issues.

vinperdom, 2021-10-17:
I'm having trouble building with scons with @bravech's branch. Below is the error message I receive. I have rosetta 2 installed, btw.

```
ld: warning: ignoring file /opt/homebrew/lib/libzmq.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: symbol(s) not found for architecture x86_64
```

jererobles, 2021-10-17:
@vinperdom are you on M1? Try setting your arch manually via flags. Checkout the troubleshooting section from tools/README.md in the PR I mentioned above.

jererobles, 2021-10-17:
@vinperdom if you have M1 then make sure to use `ARCHFLAGS="-arch arm64" ./tools/mac_setup.sh`

lukasloetkolben, 2021-10-17:
You can have try on my repo. 
I have a readme, when following the steps you should be able to build OP.
After cloning the Repository you have to make a 'git checkout 5ae276fb2ab4468736e73dd5a829493b028320df'

vinperdom, 2021-10-17:
I think I got better results with @lukasloetkolben's branch. Now I'm getting `OSError: Cannot load native module 'Crypto.Hash._MD5': Not found '_MD5.cpython-38-darwin.so'`. I'm trying to solve it but if someone already has the answer, let me know.

lukasloetkolben, 2021-10-20:
Casadi must be built from source right now.

`git clone https://github.com/casadi/casadi.git casadi
cd casadi
git checkout 3.5.5
mkdir build
cd build
cmake -DWITH_PYTHON=ON -DWITH_PYTHON3=ON ..
make -j 8
sudo make install`

smurfd, 2021-12-23:
FYI compiling casadi with clang 13 causes issues when importing
you get this error(also causes issues building openpilot on m1) : 
```
>>> import casadi
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/homebrew/lib/python3.8/site-packages/casadi/__init__.py", line 36, in <module>
    from casadi.casadi import *
  File "/opt/homebrew/lib/python3.8/site-packages/casadi/casadi.py", line 13, in <module>
    from . import _casadi
ImportError: dlopen(/opt/homebrew/lib/python3.8/site-packages/casadi/_casadi.so, 0x0002): symbol not found in flat namespace '_PyCMethod_New'

```
using clang 11, you dont get that
```
LD_LIBRARY_PATH=/opt/homebrew/Cellar/llvm@11/11.1.0_3/lib:$LD_LIBRARY_PATH
PATH=/opt/homebrew/Cellar/llvm@11/11.1.0_3/bin:$PATH
```

adeebshihadeh, 2021-12-31:
PlotJuggler support added for both M1 and x86 in #23344. Would be great if someone could test and confirm the x86 version works.

tecandrew, 2022-01-01:
> PlotJuggler support added for both M1 and x86 in #23344. Would be great if someone could test and confirm the x86 version works.

PJ works on Apple x86_86. `mac_setup.sh` needs some tweaking to get OP virtualenv setup right

```bash
((openpilot) ) ╭─atec@trabus-mba in ~/workspace/op/PlotJuggler/build on comma-master ✔ (origin/comma-master)
╰$ cmake ../
-- The C compiler identification is AppleClang 13.0.0.13000029
-- The CXX compiler identification is AppleClang 13.0.0.13000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- --------------------------------------------------
-- PlotJuggler is being built WITHOUT any ROS support
-- --------------------------------------------------
-- No build type selected, default to RelWithDebInfo
-- Found Rlog: /usr/local/include;/Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/usr/include;/Users/atec/workspace/op/PlotJuggler/3rdparty/opendbc/can;/usr/local/include
-- Found Rlog: /usr/local/lib/libcapnp.a;/usr/local/lib/libcapnpc.a;/Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/usr/lib/libbz2.tbd;/usr/local/lib/libkj.a
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: `.' is up to date.
scons: done building targets.
-- Found Cereal: /Users/atec/workspace/op/PlotJuggler/3rdparty/cereal/libcereal.a;/Users/atec/workspace/op/PlotJuggler/3rdparty/cereal/libmessaging.a;/usr/local/lib/libzmq.dylib
Cereal include dirs: /Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/usr/include
Cereal libs: /Users/atec/workspace/op/PlotJuggler/3rdparty/cereal/libcereal.a;/Users/atec/workspace/op/PlotJuggler/3rdparty/cereal/libmessaging.a;/usr/local/lib/libzmq.dylib
-- Could NOT find libdw (missing: LIBDW_LIBRARY LIBDW_INCLUDE_DIR)
-- Could NOT find libbfd (missing: LIBBFD_LIBRARY LIBBFD_INCLUDE_DIR)
-- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR LIBELF_LIBRARY LIBELF_INCLUDE_DIR)
-- Found Backward: /Users/atec/workspace/op/PlotJuggler/3rdparty/backward-cpp
-- Could NOT find MOSQUITTO (missing: MOSQUITTO_LIBRARY MOSQUITTO_INCLUDE_DIR)
[libosquittopp-dev] not found. Skipping plugin DataStreamMQTT.
QtAV_FOUND                 = FALSE
QTAV_INCLUDE_DIRS          = QTAV_INCLUDE_DIR-NOTFOUND;QTAV_INCLUDE_DIR-NOTFOUND/..
QTAV_LIBRARIES             = QTAV_LIBRARY-NOTFOUND
QTAVWIDGETS_INCLUDE_DIRS   =
QTAVWIDGETS_LIBRARIES      = QTAVWIDGETS_LIBRARY-NOTFOUND;QTAV_LIBRARY-NOTFOUND
QtAV not found. VideoPlayer plugin will not be built. Try installing it with 'sudo apt install libqtav-dev'
-- Building KissFFT with datatype: float
-- Found PythonInterp: /Users/atec/.local/share/virtualenvs/openpilot-m0sYyL2M/bin/python (found version "3.9.9")
Cereal include dirs: /Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/usr/include
Cereal libs: /Users/atec/workspace/op/PlotJuggler/3rdparty/cereal/libcereal.a;/Users/atec/workspace/op/PlotJuggler/3rdparty/cereal/libmessaging.a;/usr/local/lib/libzmq.dylib
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/atec/workspace/op/PlotJuggler/build
```

[image]

adeebshihadeh, 2022-01-07:
Both M1 and Intel should now be fully supported, largely thanks to @tecandrew!

closed 2022-01-07 by adeebshihadeh