commaai/openpilot #22576: Factory LDA warning can't be turned off

kinng, 2021-10-16:
### Describe the bug

With Factory LDA enabled and C3's default LDA warning disabled, getting factory LDA warning on the dash as well as the C3 screen every 30 seconds or so.  Have to manually move the steering wheel to get rid of it.
Attempts to disable factory LDA:
  1. With C3 powered on, failed to disable factory LDA using LDA button on steering wheel directly nor via factory menu settings and on/off switch on steering.  Both button has no affect.
  2. With C3 powered off (by pulling USB-C cable), turning off factory LDA works when using LDA switch on steering wheel or factory menu settings and on/off switch on steering wheel.  However, after booting C3, the LDA setting is turned on again, apparently by OpenPilot. 
 
So factory LDA can't be disabled with C3 and C3's LDA warning defaulted to off still won't hide the factory LDA warning resulted in the nagging warning with 30 seconds or so.

### What hardware does this issue affect?

comma three

### Which car does this affect?

Toyota Highlander Hybrid XLE 2017

### Provide a route where the issue occurs

9ff6adaf72343a2c|2021-10-15--19-59-48

### openpilot version

0.8.9

### Additional info

First time OP user with no additional hardware (such as SDSU or Pedal) installed.

cydia2020, 2021-10-17:
This shouldn't happen, the factory LDA warning messages should be blocked by panda, unless your car has a different address for `LKAS_HUD`, which I doubt it does.

Are you sure the alerts you're seeing are LDA alerts? With openpilot, LDA displays both the "hold steering wheel" icon and flashes the lane lines on the HUD, it'll also say "Lane Departure Detected". If you're seeing some other alerts generated by openpilot, (e.g. DM, comm issues, etc.) it'll manifest as a "hold steering wheel" icon on the dash.

kinng, 2021-10-17:
I did remember seeing the word LDA as well as "Hold Steering Wheel" message in amber color.  After coming back home and searching the car's manual I came across what I think was the message and icon:
[image]

A tiny nudge of the wheel will get rid of the message...though I tried squeezing the wheel everywhere , as the car manual instructed to do, without any effect.  Perhaps this indeed is an OP induced message.  This usually happens on a relatively straight road, without drifting much.