commaai/openpilot #22587: replay segfault

jererobles, 2021-10-17:
### Describe the bug

When trying to run `./replay '0c7f0c7f0c7f0c7f|2021-10-13--13-00-00' --dcam --ecam` the program outputs
```
services  std::vector(sensorEvents, gpsNMEA, deviceState, can, controlsState, pandaStates, peripheralState, radarState, roadEncodeIdx, liveTracks, sendcan, logMessage, liveCalibration, androidLog, carState, carControl, longitudinalPlan, procLog, gpsLocationExternal, ubloxGnss, clocks, ubloxRaw, liveLocationKalman, liveParameters, cameraOdometry, lateralPlan, thumbnail, carEvents, carParams, roadCameraState, driverCameraState, driverEncodeIdx, driverState, driverMonitoringState, wideRoadEncodeIdx, wideRoadCameraState, modelV2, managerState, uploaderState, testJoystick)
load route "0c7f0c7f0c7f0c7f|2021-10-13--13-00-00" with 4 valid segments
seeking to 0
downloading segment 0 ...
downloading segment 1 ...
downloading segment 2 ...
[1]    92375 segmentation fault  ./replay '0c7f0c7f0c7f0c7f|2021-10-13--13-00-00' --dcam --ecam
```
and exits. Full report here https://pastebin.com/vi4YU5yh 

### OS Version

macOS 12.0 (21A5552a)

### openpilot version or commit

7b5d915

### Additional info

Pulled latest master is hopes #22556 fixed it but no dice

pd0wm, 2021-10-17:
Thanks for the full crash log! Looks like a crash in `libcurl` downloading the data:

```
Thread 14 Crashed:: QThread
0   libcurl.4.dylib               	    0x7ff821e4eb05 curl_multi_info_read + 24
1   replay                        	       0x10de1957c httpMultiPartDownload(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, int, std::__1::atomic<bool>*) + 1676 (util.cc:85)
2   replay                        	       0x10de17a1b Segment::downloadFile(QString const&)::$_3::operator()() const + 163 (route.cc:155) [inlined]
3   replay                        	       0x10de17a1b decltype(std::__1::forward<Segment::downloadFile(QString const&)::$_3>(fp)()) std::__1::__invoke<Segment::downloadFile(QString const&)::$_3>(Segment::downloadFile(QString const&)::$_3&&) + 163 (type_traits:3747) [inlined]
4   replay                        	       0x10de17a1b std::__1::invoke_result<Segment::downloadFile(QString const&)::$_3>::type std::__1::invoke<Segment::downloadFile(QString const&)::$_3>(Segment::downloadFile(QString const&)::$_3&&) + 163 (functional:2984) [inlined]
5   replay                        	       0x10de17a1b void QThread* QThread::create<Segment::downloadFile(QString const&)::$_3>(Segment::downloadFile(QString const&)::$_3&&)::'lambda'(auto&&...)::operator()<>(auto&&...) + 163 (qthread.h:180) [inlined]
6   replay                        	       0x10de17a1b decltype(std::__1::forward<Segment::downloadFile(QString const&)::$_3>(fp)()) std::__1::__invoke<QThread* QThread::create<Segment::downloadFile(QString const&)::$_3>(Segment::downloadFile(QString const&)::$_3&&)::'lambda'(auto&&...)>(Segment::downloadFile(QString const&)::$_3&&) + 163 (type_traits:3747) [inlined]
7   replay                        	       0x10de17a1b void std::__1::__async_func<QThread* QThread::create<Segment::downloadFile(QString const&)::$_3>(Segment::downloadFile(QString const&)::$_3&&)::'lambda'(auto&&...)>::__execute<>(std::__1::__tuple_indices<>) + 163 (future:2186) [inlined]
8   replay                        	       0x10de17a1b std::__1::__async_func<QThread* QThread::create<Segment::downloadFile(QString const&)::$_3>(Segment::downloadFile(QString const&)::$_3&&)::'lambda'(auto&&...)>::operator()() + 203 (future:2179)
9   replay                        	       0x10de181fa std::__1::__deferred_assoc_state<void, std::__1::__async_func<QThread* QThread::create<Segment::downloadFile(QString const&)::$_3>(Segment::downloadFile(QString const&)::$_3&&)::'lambda'(auto&&...)> >::__execute() + 26 (future:906)
10  libc++.1.dylib                	    0x7ff80d6ef68e std::__1::__assoc_sub_state::copy() + 46
11  libc++.1.dylib                	    0x7ff80d6ef899 std::__1::future<void>::get() + 25
12  QtCore                        	       0x11218bfe9 0x11216c000 + 131049
13  libsystem_pthread.dylib       	    0x7ff80d78f514 _pthread_start + 125
14  libsystem_pthread.dylib       	    0x7ff80d78b02f thread_start + 15
```

deanlee, 2021-10-17:
@jererobles : thanks for the log. but this crash reports can't help much, it's from an old version  (there is no function Segment::downloadFile in 7b5d915, it  has been removed two days ago)

closed 2021-10-20 by jererobles