commaai/openpilot #22597: Match button engagement logic in panda/openpilot

pd0wm, 2021-10-18:
Openpilot tries to engage on the falling edge of the set/resume button. Panda safety modes (at least for Honda/Hyundai) sets `controls_allowed` while the button is pressed. This can cause a mismatch when the brake and the set button are released at the same time. In this case the panda never sets `controls_allowed`, but openpilot does try to engage because the brake is no longer pressed when the falling edge is triggered one cycle later.

pd0wm, 2022-03-07:
Drive where this happens with the gas: `fbbb958f3157f063|2022-03-03--21-13-55--7`

adeebshihadeh, 2022-03-16:
Hondas done in https://github.com/commaai/panda/pull/874/

adeebshihadeh, 2022-03-16:
Hyundai Longitudinal done in https://github.com/commaai/panda/pull/875
GM done in https://github.com/commaai/panda/pull/876

PR #23975 (2022-03-16): Fix controls mismatch on button enable cars + test against panda safety
resolves #22597

Fixes a controls mismatch on Honda Nidecs with pedals, Honda Bosch longitudinal, Hyundai longitudinal, and GM cars.
closed 2022-03-16 by adeebshihadeh