commaai/openpilot #22746: Mazda: better alert when LKAS is manually disabled

adeebshihadeh, 2021-10-30:
The `LANE_LINES` signal in `CAM_LANEINFO` can be used to determine this. Currently, this shows up as an EPS fault, but we can show a more actionable alert in this case.

BBBmau, 2021-11-25:
When looking at how to resolve this issue. I found where exactly to use the LANE_LINES signal, but when it comes to wanting to show an error that isn't *EPS Fault" are you referring to showing it on the comma device itself? Wasn't aware that it showed these kinds of errors on the device.

Jafaral, 2021-11-25:
@adeebshihadeh, do you want to add a new alert or just override the stock setting so that lkas is always enabled when using OP?

adeebshihadeh, 2021-11-25:
Let’s make a new alert for this

Jafaral, 2021-11-25:
Good, I prefer that route; being consistent with the stock system and what the driver expect in this case when they press the button to manually disable lkas.

Isn't this something other cars have as well? At least some of them? Regardless of having a physical button or just an option in car settings

adeebshihadeh, 2021-12-15:
> Isn't this something other cars have as well? At least some of them? Regardless of having a physical button or just an option in car settings

I haven't seen this on other cars.

This should be pretty simple to implement. Just add a new EventName in cereal/car.capnp like `lkasDisabled`, make an alert in `selfdrive/controls/lib/events.py`, and add the event in Mazda's car interface.

PR #23273 (2021-12-20): Mazda: alert when LKAS is disabled
Fixes: #22746
Depends: https://github.com/commaai/cereal/pull/235
closed 2021-12-20 by adeebshihadeh