commaai/openpilot #22775: 2019 Kia Sorento: e2e_long not stopping for stop lights/signs

kalub92, 2021-11-02:
### Describe the bug

I switched to the `e2e_long` branch and enabled OP long by setting `DisableRadar_Allow` to `1` and while driving, the yellow triangle appeared behind a lead car so I’m fairly certain I enabled it correctly. When starting the car there was a FCW and cruise fault alert on the instrument cluster but that went away after a few seconds and I could then press the Cruise button on my steering wheel and engage OP with no issue.

I went for a drive and noticed the car slowed down behind a lead car in a much smoother way than with radar, but when approaching red lights or stop signs the car would not slow down or stop. I thought I wasn’t waiting long enough for E2E long to see the light and respond, but it didn’t slow to stop even once for many stop lights.

Additionally, it wasn’t slowing for curves.

Another weird thing was when I would engage OP, the max speed was 25 MPH when my car usually has a minimum of 20 MPH. My instrument cluster would say 20 but the C3 would say 25.

I could also engage OP from 0 MPH (can’t usually) but it would throw an alert saying “steering unavailable” then go into “TAKE CONTROL IMMEDIATELY” mode for a few seconds. It would do the same thing if I attempted to resume OP (vs. hitting the set button) below 20 MPH.

There was another time where I disengaged but the car started accelerating. Kinda freaky.

### What hardware does this issue affect?

comma three

### Which car does this affect?

Kia Sorento 2019

### Provide a route where the issue occurs

b78a19b557b991ab|2021-11-01--19-40-10

### openpilot version

0.8.11

pd0wm, 2021-11-02:
Regarding your car specific issues. You tried to run longitudinal control on a car where that's not supported at all. You failed to mention that you also made local changes to `selfdrive/car/hyundai/interface.py` to add your car to the list of cars that support longitudinal control. However, the reason that your car (Sorento) is not in there is because it uses the Hyundai Legacy safety mode which does not support longitudinal control.

You got into a weird state where openpilot thought it was controlling longitudinal, but all longitudinal related messages where blocked by the panda, while actually your car's ACC was controlling longitudinal. This caused things like disengage on gas not to work, since openpilot did not cancel the stock ACC.

Making bug reports about things like this not working costs us quite a bit of time. We look into it because the issues sound serious, but then it turns out openpilot is not even supposed to work in this situations, and you caused the issue yourself.

HaraldSchafer, 2021-11-02:
Additionally we generally only accept bug reports for things that affect upstream openpilot. E2e long is an experimental dev branch, it's not guaranteed to do anything.

closed 2021-11-02 by pd0wm