commaai/openpilot #22802: Look into using the VIN query for fingerprint caching

adeebshihadeh, 2021-11-05:
Quicker fingerprinting times will become more important as more queries are added to support FW fingerprinting for all cars and as startup times go down. Right now, we fingerprint on the first drive after booting, then [cache](https://github.com/commaai/openpilot/blob/6bef60ddf56d49a2d6afdf1dfcb85cc3da8211a6/selfdrive/car/car_helpers.py#L101) that for subsequent drives but discard the cache on boot. Instead, we'd like to keep the cache around until the VIN changes.

gregjhogan, 2021-11-05:
An alternative is send tester present to all ecus and the list of addrs that respond should tell you the vehicle manufacturer (then only run fingerprinting for a single manufacturer).  Detecting VIN change is simpler, but firmware changes will no longer be detected (ensuring the interface to the car had not changed).

pd0wm, 2021-11-06:
Think this came up in the meeting as well, but we have to be careful to not cache these results forever. Any changes/breakage in the fingerprinting code wouldn't be discovered for quite a while.

adeebshihadeh, 2021-12-15:
Closing as this doesn't need to be an issue. This is already the obvious thing to look into once fingerprinting is enough of a concern for startup time.

closed 2021-12-15 by adeebshihadeh