commaai/openpilot #22821: Disable reboot/shutdown buttons while onroad

pd0wm, 2021-11-08:


Achilles308, 2021-11-08:
Maybe gated for certain cars? On my Toyota Corolla Hatchback these buttons work fine if you're smart about when you use them (disengage OP then reboot).

I think turn off should always be avaliable, personally. Maybe that means some kind of warning to the user to take over control of the car before it does shutdown.

Also, if these are removed, then please add some way to change toggles and implement them without requiring a reboot. Like a soft reboot?

pd0wm, 2021-11-09:
Decided to change this a little since blocking the buttons would just make people unplug their device. We allow reboot/shutdown when not engaged. The actual shutdown/reboot will be moved to manager so it can first do a graceful shutdown of all the openpilot services and close the relay to hand control back to the car.

PR #22882 (2021-11-12): Move shutdown/reboot handling to manager
Closes #22821

We should probably add something to boardd to put the panda back into elm/no output mode when it's shut down cleanly. Think that should go in a separate PR.

[image]
[image]
closed 2021-11-12 by adeebshihadeh