commaai/openpilot #22822: Increase the usage of augmented assignment statements

elfring, 2021-11-08:
:eyes: Some source code analysis tools can help to find opportunities for improving software components.
:thought_balloon: I propose to [increase the usage of augmented assignment statements](https://docs.python.org/3/reference/simple_stmts.html#augmented-assignment-statements "Augmented assignment statements") accordingly.

Would you like to integrate anything from a transformation result which can be generated by a command like the following?
(:point_right: Please check also for questionable change suggestions because of an evolving search pattern.)
```
lokal> perl -p -i.orig -0777 -e 's/^(?<indentation>\s+)(?<target>\S+)\s*=\s*\k<target>[ \t]*(?<operator>[+\-%&|^@]|\*\*?|\/\/?|<<|>>)/$+{indentation}$+{target} $+{operator}=/gm' $(find ~/Aktivitäten/openpilot/lokal -name '*.py')
```

:crystal_ball: How will the development interests evolve further also according to update candidates in 22 lines of this software?

pd0wm, 2021-11-08:
Not interested

closed 2021-11-08 by pd0wm

elfring, 2021-11-09:
:thought_balloon: Do any factors hinder the wider application of [functionality which became generally available with Python 2](https://docs.python.org/3/whatsnew/2.0.html#augmented-assignment "Augmented assignments")?