commaai/openpilot #22823: reserved identifier violation

elfring, 2021-11-08:
### Describe the bug

I would like to point out that identifiers like “[`m__raw_measurements`](https://github.com/commaai/openpilot/blob/13a6e3c68f94d39890602acd6955ac9025e34498/selfdrive/locationd/generated/ubx.h#L116 "Update candidate")” and “[`p__io`](https://github.com/commaai/openpilot/blob/13a6e3c68f94d39890602acd6955ac9025e34498/selfdrive/locationd/generated/gps.cpp#L66 "Another update candidate")” [do eventually not fit](https://www.securecoding.cert.org/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

### What hardware does this issue affect?

comma three, comma two, EON Gold

### Provide a route where the issue occurs

77611a1fac303767|2020-05-11--16-37-07

### openpilot version

25c8baae41ddeb6070597e3f27a48595ac8319ca

pd0wm, 2021-11-08:
This is auto-generated code from kaitai. So you should probably make an issue on their repo: https://github.com/kaitai-io/kaitai_struct/issues

elfring, 2021-11-08:
:thought_balloon: Do you control if such a parser generator would be applied for an implementation of the programming language “C++”?

pd0wm, 2021-11-08:
Not sure what you mean by that? Nothing is currently broken, so don't think this worth an issue. Feel free to submit a PR to the kaitai generator to conform to the standard.

closed 2021-11-08 by pd0wm

elfring, 2021-11-09:
>Nothing is currently broken,

This view depends on the interpretation of the handling for mentioned identifiers.
Can [the use of double underscores](https://stackoverflow.com/questions/228783/what-are-the-rules-about-using-an-underscore-in-a-c-identifier#answer-228797 "What are the rules about using an underscore in a C++ identifier?") be treated according to naming rules of the programming language “C”?

>so don't think this worth an issue.

I imagine that contents from a file like “[gps.cpp](https://github.com/commaai/openpilot/blob/13a6e3c68f94d39890602acd6955ac9025e34498/selfdrive/locationd/generated/gps.cpp#L66 "Update candidate: subframe_1_t constructor")” can trigger further software development considerations.
How do you think about to avoid that this software [depends on undefined behaviour](https://en.cppreference.com/w/cpp/language/identifiers#In_declarations "Description for C++ identifiers")? :thinking: