commaai/openpilot #22824: Remove unnecessary null pointer checks

elfring, 2021-11-08:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [CameraBuf](https://github.com/commaai/openpilot/blob/5246f0231e6bfa572c0dd89c60e538910817faa2/selfdrive/camerad/cameras/camera_common.cc#L131 "Destructor for CameraBuf")
- [gps_t::_clean_up](https://github.com/commaai/openpilot/blob/13a6e3c68f94d39890602acd6955ac9025e34498/selfdrive/locationd/generated/gps.cpp#L52 "Function gps_t::_clean_up")
- [ubx_t::_clean_up](https://github.com/commaai/openpilot/blob/13a6e3c68f94d39890602acd6955ac9025e34498/selfdrive/locationd/generated/ubx.cpp#L62 "Function ubx_t::_clean_up")

pd0wm, 2021-11-08:
Feel free to open a PR to remove them from non autogenerated code.

closed 2021-11-08 by pd0wm

elfring, 2021-11-09:
:thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability-delete-null-pointer.html "Deletion of unnecessary null pointer checks")?