commaai/openpilot #22836: Vehicle can't keep in lane on right bank turn on highway, drifted half a car into passing lane

VenomrushCT, 2021-11-09:
### Describe the bug

Upgraded to v0.8.10 and notice vehicle is unable to keep in lane during a right bank turn on a highway, it would drift to the passing lane, at times half a car length. 

I did not notice this issue in earlier version on the same route.

Route: f72a3ad7dc38d5cf|2021-11-06--09-18-49--19
Happened in an earlier right bank turn but less severe: f72a3ad7dc38d5cf|2021-11-06--09-18-49--17

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

f72a3ad7dc38d5cf|2021-11-06--09-18-49--19

### openpilot version

0.8.10

Verylukyguy, 2021-11-09:
I have noticed this too.  The vehicle seems to make left turns on the freeway okay and stays comfortably away from the right lane line, but, it crowds, hugs, and even drives on or over the left lane line when making right turns.

HaraldSchafer, 2021-11-09:
It just looks like it's hitting torque limits to me
[image]

VenomrushCT, 2021-11-09:
I never had this issue in earlier version. It was perfect through this exact turn. I did not have to intervene. 

My vehicle is a Honda CR-V 2018 and currently using laneless. Unless something else vehicle specific was changed. 

I will try to turn off laneless and observe the behavior tonight and report.

HaraldSchafer, 2021-11-10:
It would be helpful if you could show me a segment doing this turn correctly. Do you have a segment id of it working well previously?

VenomrushCT, 2021-11-10:
Unfortunately I cancelled Prime so don't have the old drives. 

I went out again just now to take that turn twice. At the same speed (70mph) replicate the same condition as best I can. This is what I observed:

f72a3ad7dc38d5cf|2021-11-09--19-14-13--10 in this segment I turned off laneless. Seems to take the turn more confident. 

f72a3ad7dc38d5cf|2021-11-09--19-33-10--5 in this segment I reset the calibration and turned on laneless again. Seems to take the turn more confident. 

Another unrelated issue I can't find the Nov 6 route morning anymore. It hasn't been 2 weeks?

HaraldSchafer, 2021-11-10:
I can still see it?

https://connect.comma.ai/f72a3ad7dc38d5cf/1636204730142/1636206418650

That doesn't work for you?

VenomrushCT, 2021-11-10:
> I can still see it?
> 
> https://connect.comma.ai/f72a3ad7dc38d5cf/1636204730142/1636206418650
> 
> That doesn't work for you?

It's not accessible for me anymore 

**User Admin**
[image]

**Connect**
[image]

HaraldSchafer, 2021-11-11:
Ah looks like drives are only accessible for 3 days without prime :/

VenomrushCT, 2021-11-11:
So is this issue valid? I am not able to view the old drive to visually see. 

Left turns on highway perform amazing, issue seems to be with the right turn. 
I took the same road in the opposite direction, so same amount of turning required, never had any warnings for left turns.

HaraldSchafer, 2021-11-11:
From the data I saw, it looked like it was hitting the torque limit. It's possible some changes in how the model plans the path makes it take the turn worse with the same amount of torque, but not something I can say conclusively from the data.

If you have more examples of it behaving very poorly again, post them here.

Achilles308, 2021-11-12:
> From the data I saw, it looked like it was hitting the torque limit. It's possible some changes in how the model plans the path makes it take the turn worse with the same amount of torque, but not [...]

Just anecdotally, I have noticed on banked turns with laneless 0.8.11 does better at not cutting the corner early. In other words, it no longer cheats the inside of the curve to be able to make the turn. Which, for a turn that was right on that limit before, it would fail to maintain the lane now.

For my car, I'm getting more alerts than before in these types of banked turns, however, the turns are completed much smoother and with less center line cutting. So damned if you do, damned if you don't. I would rather it not cut the double yellow entering the turn and fail to maintain lane due to low torque than to constituently cut the yellow for turns it could have made without crossing the yellow ever.

closed 2021-11-30 by adeebshihadeh

VenomrushCT, 2021-11-30:
This is an issue, why is it closed?

adeebshihadeh, 2021-11-30:
The conclusion with the given information was it's probably hitting the torque limit. More information was requested over two weeks ago with none given. If you'd like us to look into this more, post further examples.