commaai/openpilot #22837: Loggerd: 100% CPU- should not use bz2 compression on camera video

jwolffe, 2021-11-09:
### Describe the bug

When simulating with Carla, I observed 100% usage of the CPU core that loggerd runs on. The roadCameraState is being handled through the logger_log routine that uses bz2 compression. The video is not compressible by that lossless method, and the high data rate (~60MB/s) causes loggerd to be overloaded.

### OS Version

Ubuntu20.04LTS

### openpilot version or commit

53e095343de56ff838f36ed9feda9f31ff6358b6

### Additional info

I tried fixing it by changing line 334 of loggerd.cc to:
     if (!it.should_log || strstr(it.name,"CameraState")) continue;

This prevents any of roadCameraState, driverCameraState, or wideCameraState from being logged through the bz2 path, and seems to solve the immediate problem. The camera video is compressed to .hvec by the encoder threads. There is a question of how this issue should be fixed, whether simply applying my one-liner to loggerd.cc, or perhaps some sort of system-level solution that coordinates the should_log boolean from services.py together with the .hvec logging done by the encoder threads.

adeebshihadeh, 2021-11-09:
We'd like to keep our issue tracker focused on comma two/three, so I'm gonna close this. loggerd for PC mostly exists for allowing the tests to run on PC. We'd certainly merge PRs to improve the PC experience though!

As for the frames being the in cameraState packets, we should fix this. CARLA uses a fake camera in camerad called "frame stream" as it was easier to stuff the frames in the capnp messages than send over visionipc from python as camerad normally does. However, we now have a nice python visionipc server implementation, so we can do this properly now. I started that in #21324 but didn't have time to finish it. Feel free to take that PR and finish it up.

closed 2021-11-09 by adeebshihadeh

jwolffe, 2021-11-09:
First, my apologies, I didn't intend to misuse the "Issues". I'm still learning how things are done in this community. 

I will look into that PR #21324 and see what I can figure out. It is a little difficult for me to see how things work- I only have the PC simulation environment, and I don't have much understanding of the "big picture" yet. I've just been finding and fixing bugs in the simulation.

So, the cereal implementation supports the ...CameraState messages (and loggerd erroneously applies bz2 compression), but that was just to provide a temporary "hack" for the PC simulation. Am I correct that they should be removed from the cereal code when the PR gets finished, or is something else intended?

Thank you for the guidance.

adeebshihadeh, 2021-11-09:
No worries :)

Check out our new docs at https://docs.comma.ai. Fortunately, fixing the simulator should be a great way to learn the codebase. Basically, loggerd indiscriminately subscribes to all the services defined in [cereal/services.py](https://github.com/commaai/cereal/blob/master/services.py) that specify the `should_log` flag.

Here's a more in-depth overview:

`camerad` captures frames at 20Hz and for every frame. it sends out the frame over visionipc and the `xxxCameraState` metadata packet over msgq. When the CARLA bridge was initially written, there wasn't an easy way to send frames over visionipc from python, so an `image` field was added to the [`FrameData` struct](https://github.com/commaai/cereal/blob/master/log.capnp#L151). The ["frame stream" camera](https://github.com/commaai/openpilot/blob/master/selfdrive/camerad/cameras/camera_frame_stream.cc) was written to have camerad listen for the `xxxCameraState` packets instead of sending them out like it normally does, then send out the image stuffed in the `image` field over visionipc.

jwolffe, 2021-11-09:
Thank you, that helps a lot. So, the intent is to keep logging the CameraState messages for the metadata, but removing the video image stream from Carla and instead get it into openpilot using visionipc (and log it normally through the encoder threads). Did I get that right?

adeebshihadeh, 2021-11-09:
I think you've got the right idea, but didn't quite understand what you said. The goal is to send the frames over visionipc, which allows us to stop stuffing them in roadCameraState. In this case, we also  wouldn't run camerad at all for CARLA, as the bridge would handle all of its functionality. The encoders currently work correctly as camerad's frame stream takes care of sending the frames over visionipc (they're just in two places now - visionipc and roadCameraState).

jwolffe, 2021-11-10:
Sorry if I wasn't clear, but I think we are on the same page now. Yes, it makes sense that the bridge would take on the role of camera, sending a video stream to visionipc and also publishing metadata in roadCameraState. Thank you!