commaai/openpilot #22854: unplugging my  DSU on my 2018 Toyota Prius

bugbitbyte, 2021-11-10:
### Describe the bug

Openpilot has always worked with my 2018 Toyota Prius On comma 2 and comma 3.  But one day my comma three did a camera malfunction and I had to factory reset the comma.  After I did that My comma did not want to work cuz the DSU was unplugged so I turn my car off and plugged in the DSU and I get assistance but it's not open pilot. Now every time I unplug the DSU, cruise control does not work aka openpilot.  I only have adaptive cruise control with the DSU plugged in but it's not openpilot.  I also get the yellow/orange pre-collision system  light flashing when the DSU is unplugged. 

### What hardware does this issue affect?

comma three, comma two

### Which car does this affect?

2018

### Provide a route where the issue occurs

?

### openpilot version

0.8.10

### Additional info

Please 🙏 help, CommaAi team told me someone Eric from here can help.

adeebshihadeh, 2021-11-10:
Check out our [Discord](https://discord.comma.ai) or GitHub Discussions for getting started.

closed 2021-11-10 by adeebshihadeh

bugbitbyte, 2021-11-10:
I checked out the page, how come you close this on me instantly

bugbitbyte, 2021-11-10:
I checked out the page still need help