commaai/openpilot #22866: Missing Sconscript file under rednose/

HaotianRen12, 2021-11-11:
### Describe the bug

Hi, I was trying to run "$ scons -c" under openpilot/ and there was an error saying " Missing SConscript 'rednose/SConscript' ". I could find an SConscript file under rednose/rednose/, but not under rednose/. I'm new to openpilot and not sure if there is supposed to be that file under rednose/, any help would be appreciated!

### OS Version

Ubuntu 20.04

### openpilot version or commit

aa38a5fe34008718ea53ffa339f375ac0b3990a1

abdulrahman004, 2021-11-11:
did you do `git clone --recurse-submodules https://github.com/commaai/openpilot.git` before trying to build.
rednose is a git submodule that needs lives outside opnepilot repo

HaotianRen12, 2021-11-11:
Thank you so much, it solved my issue. I didn't add "--recurse-submodules" and cloned rednose manually from git.

closed 2021-11-11 by HaotianRen12