commaai/openpilot #22888: sensord : Gyro values are unreasonable sometimes during init

nuwandavek, 2021-11-12:
### Describe the bug

When initialized, sometimes the gyro values are unreasonable. We could potentially filter these out in sensord.

[image]

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

fbbb958f3157f063|2021-11-12--11-15-20/0

### openpilot version

c7be73b826b21947197f24c6a60fc5a7098e42af

adeebshihadeh, 2021-11-12:
How many values are nonsense? Is it just the first one?

adeebshihadeh, 2022-08-14:
@nuwandavek is this still an issue?

nuwandavek, 2022-09-09:
This happens in ~25% of first segments (checked in 10k routes). This is not a problem as there are lots of checks downstream, but good to resolve it in `sensord`! May have to skip the first 5-10 readings. 
[image]
[image]

adeebshihadeh, 2022-09-09:
@gast04 let's write a unit test to catch this and fix it

incognitojam, 2022-09-30:
@gast04 can we close this?

closed 2022-09-30 by adeebshihadeh