commaai/openpilot #22903: Cruise cancel malfunctions

adeebshihadeh, 2021-11-14:
The new cruise cancellation alert has caught some existing car bugs:

Volt
```
qadmus yesterday
Contributor

GM Volt hits this event upon engage
```

Toyota
```
This creates an annoying issue on Toyotas whereby if the cruise is cancelled by the brakes while the car is in standstill, it throws an alert.

https://youtube.com/shorts/A9rEyBFc_-g
```

DarkSlaayer, 2021-11-14:
2020 Hyundai Sonata gets this immediately after engaging.

openpilot failed to cancel cruise

cydia2020, 2021-11-15:
> This creates an annoying issue on Toyotas whereby if the cruise is cancelled by the brakes while the car is in standstill, it throws an alert.

https://youtube.com/shorts/A9rEyBFc_-g

I shall add something to my comment: (this applies to Toyota and Lexus only ofc)

I'd assume that you know the infamous "ding" sound and dash faults associated with pcm_cancel on Toyotas, as documented by #21708 .

The "openpilot failed to cancel cruise" alert only shows if the dash faults are created.

The dash fault only appears if openpilot cancels cruise before the car cancels it (E.g. on driver brake press when going downhills or stopping (more details are provided later in this comment), gas press, or faults).

These two facts combined led me to believe that there is a delay (roughly 2 seconds) in the CRUISE_ACTIVE message. I have attempted to use the CRUISE_STATE message as a cruise control state indication, however it resulted in a similar behaviour.

I'd also like to say after testing the stock cruise system for its behaviours, I suspect that this is not a car bug, but in fact, a safety feature implemented by Toyota, if a driver was to use the stock system, and the system loses track of the front car while braking, the ACC will disengage while displaying "no close lead car", 4 chimes will be played from the dash, and the vehicle will come to a complete stop just behind the last know position of the lead car without the cruise being active; it also looks like it applies to brake presses while stock ACC is slowing the car down, the stock system will keep braking actions for a while (without ACC engaged, and regardless if the driver had let go of the brake pedal), then release the brakes after about 2s, allowing full control from the driver afterwards.

I'll report back if I observe anything else, for now I'd suggest comma revert this change, or at least minimise the alert, until a solution has been found.

adeebshihadeh, 2021-11-16:
@DarkSlaayer can you post a route and make sure it's fully uploaded in useradmin?

adeebshihadeh, 2021-11-16:
I removed the alert for this release, but left the event in for logging. Once 0.8.11 goes out, we should be able to find all the cases where this triggers and re-enable the alert for 0.8.12.

ffhspa, 2021-11-16:
cruiseMismatch hHappens on Honda Civic Bosch Diesel Car as well on Op 0.8.11

jnadke, 2021-11-17:
I don't know if this is related, on Chrysler this breaks cruise control entirely.

On 0.8.10 OP stock I get "Cruise Control Unavailable"

VirtuallyChris, 2021-11-18:
Got it in the Accord `53dde38586d5978c|2021-11-16--12-44-44--19`

maniloko93, 2021-12-13:
I get it on Honda Civic 2020 as well  0.8.11

closed 2022-08-13 by adeebshihadeh