commaai/openpilot #22936: Cruise control unavailable error

jnadke, 2021-11-17:
### Describe the bug

Using OpenPilot 0.8.10 official release results in "Cruise Control Unavailable" on Chrysler Pacifica and completely breaks cruise control for the entire drive.

I last used full stock on 0.8.7 and it did not do this.

### What hardware does this issue affect?

comma two

### Which car does this affect?

2021 Chrysler Pacifica Hybrid

### Provide a route where the issue occurs

ac2d665ad9da11e6|2021-11-15--17-31-48

### openpilot version

0.8.10

vanillagorillaa, 2021-11-19:
Can you provide more details on this? What time in the route does this occur? Does it occur after a certain button press? Does it occur right at startup? Is this error displayed on your comma two, or is it displayed on the vehicle?

jnadke, 2021-11-19:
@vanillagorillaa It happens at 17:39:08

Pretty much immediately as I enabled Cruise Control to activate OpenPilot.

Immediately the vehicle started slowing down (you will see a decrease in speed in the data) until I realized 0.8.10 was breaking cruise control.

jnadke, 2021-11-19:
On Chrysler Vehicles the ACC/LKAS system has a separate toggle for on/off than the ACC Set speed.

It doesn't break when the ACC/LKAS system is activated.   Instead it appears to break when the set speed button is used (which activates OpenPilot / LKAS).

OpenPilot still displayed a green box thinking it was in control of the vehicle.  No error was displayed.

vanillagorillaa, 2021-11-19:
IMHO finding the stock release version where this begins to occur would certainly help narrow down your issue to a single release.

BrianWisniewski, 2021-11-21:
Video of the issue would likely help get this resolved faster.

jnadke, 2021-11-21:
I will attempt to check if 0.8.9 or 0.8.8 broke it and get video of how it fails later today.

jnadke, 2021-11-21:
To elaborate on why I opened this issue and checked 0.8.10 stock, is because while 0.8.10 stock is completely broken, the community fork jvepilot 0.8.10 functions but gets a high incidence of LKAS system errors.

So possibly something got added in 0.8.10 the system doesn't agree with.

vanillagorillaa, 2021-12-03:
@jnadke Did you check stock versions of 0.8.9 and 0.8.8 to see if the issue was there or if it is indeed in 0.8.10?

jnadke, 2021-12-07:
@vanillagorillaa I apologize the pacifica is not my daily driver, I'm loading the old versions now to test them.

jnadke, 2021-12-16:
@vanillagorillaa I'm going to recommend closing this issue for now.

I could not reproduce it when I went to 0.8.10+ again.  I don't know what caused the issue but I am unable to reproduce it now.

closed 2021-12-16 by jnadke