commaai/openpilot #22985: Recently around 50% time, the device always say "LKAS Fault, restart the car to engage."

visionandy, 2021-11-20:
### Describe the bug

I have talked with the customer service during the first month since I bought the device, the customer service sent me a new usb-c cable, but the problem is still on until today. It's annoying recently, it becomes more and more frequent than before, at least 50% or 70%, it always show "LKAS Fault, restart the car to engage." 
My car model is 2017 Jeep Grand Cherokee. Thanks for the help in advance. 

### What hardware does this issue affect?

comma two

### Which car does this affect?

2017 Jeep Grand Cherokee.

### Provide a route where the issue occurs

0844505b5a3fc8d8|2021-11-19--17-57-19	

### openpilot version

recent version, just updated.

### Additional info

dongle_id: 0844505b5a3fc8d8

adeebshihadeh, 2021-11-21:
Probably similar as #22936. @vanillagorillaa want to confirm?

wangstein, 2021-12-03:
I am having the same problem. 2017 Jeep Grand Cherokee. It’s either. It recognized or dashcam mode. Very annoying.

vanillagorillaa, 2021-12-03:
@wangstein What is your device dongleID?

wangstein, 2021-12-03:
2a6e00ad5183c2d6

wangstein, 2021-12-03:
@vanillagorillaa I’m seeing platform as 
JEEP GRAND CHEROKEE V6 2018 but my car is 2017.

vanillagorillaa, 2021-12-03:
This is fine, the platform is the same between the years. This isn't the issue.

wangstein, 2021-12-03:
Ok. Do u know if this is hardware or software. Can it be fixed? Got a big road trip coming up love to use it. Any pointers?

mogorman, 2021-12-07:
I started having this problem as well after upgrading from 0.8.9 -> 0.8.10 as well

wangstein, 2021-12-07:
@mogorman I'm not sure if @vanillagorillaa did something on the backend... but since i sent in the dongle id 3days, i been having less of the issue. So thank you @vanillagorillaa if you did something magical! :)

adeebshihadeh, 2022-01-27:
This should be fixed in 0.8.13. You can install master-ci with `openpilot-nightly.comma.ai` or wait until 0.8.13 goes to release. Let us know if you still have issues.

closed 2022-01-27 by adeebshihadeh

visionandy, 2022-05-08:
Hi @adeebshihadeh, good morning! Thanks for the help. This problem starts to occur more often recently, especially in yesterday's trip. I can upload more information when the video is uploaded to the cloud.