commaai/openpilot #23006: C3: full rebuild gets OOM killed

adeebshihadeh, 2021-11-23:
SCons gets killed on a full rebuild on a C3 due to it running out of memory. We can lower the number of jobs, but this will unnecessarily make a lot of builds slower. @deanlee want to look into this?

deanlee, 2021-11-23:
```
Object counts:
       pre-   post-    pre-   post-   
       read    read   build   build   Class
          0     639     639    2963   Action.CommandAction
          0      24      24      24   Action.CommandGeneratorAction
          0     102     102     102   Action.FunctionAction
          0      78      78      78   Action.LazyAction
          0      12      12      12   Action.ListAction
          0     202     202     202   Builder.BuilderBase
          0      22      22      22   Builder.CompositeBuilder
          0     574     574     574   Builder.OverrideWarner
          0       4       4       4   Environment.Base
          0      10      10      10   Environment.EnvironmentClone
          0     355     355     800   Environment.OverrideEnvironment
          0     955     955    1605   Executor.Executor
          0       0       0    1894   Executor.Null
          0    1032    1032    3211   Node.FS.Base
          0      85      85     735   Node.FS.Dir
          0     339     339    1868   Node.FS.File
          0    1348    1348    3527   Node.Node
```
There are many nodes and executors. one possible way is to clear unnecessary #include. But I'm not sure how much it can help

deanlee, 2021-11-23:
It seems to be helpful. for example, if we remove QApplication,QScreen from qt_window.h, we can save 4 nodes and executors, and the memory usage reduced from 1715453952 to 1698000896
```
Memory after building targets:            1698000896
Object counts:
       pre-   post-    pre-   post-   
       read    read   build   build   Class
          0     955     955    1601   Executor.Executor
          0       0       0    1894   Executor.Null
          0    1032    1032    3207   Node.FS.Base
          0      85      85     731   Node.FS.Dir
          0     339     339    1868   Node.FS.File
          0    1348    1348    3523   Node.Node
```

pd0wm, 2021-11-23:
@deanlee Why are you testing with -j6? We build with nproc - 1, so that would be 7.

It can't seem to reproduce this reliably, but noticed building the locationd objects/filters seem to use a lot of RAM, over 30% peak for the larger ones (loc_4). It probably runs out of ram if the timing works out in such a way that multiple are built at the same time.

Shall we tweak the dependencies to ensure only one of the expensive locationd objects are built at the same time?

pd0wm, 2021-11-23:
Output after the PR (with -j6 for comparison):

```
scons: done building targets.
Memory before reading SConscript files:     24268800
Memory after reading SConscript files:     351543296
Memory before building targets:            755793920
Memory after building targets:             811094016
Object counts:
       pre-   post-    pre-   post-   
       read    read   build   build   Class
          0     646     646    2854   Action.CommandAction
          0      20      20      20   Action.CommandGeneratorAction
          0     101     101     101   Action.FunctionAction
          0      78      78      78   Action.LazyAction
          0      15      15      15   Action.ListAction
          0     203     203     203   Builder.BuilderBase
          0      18      18      18   Builder.CompositeBuilder
          0     539     539     539   Builder.OverrideWarner
          0       5       5       5   Environment.Base
          0      10      10      10   Environment.EnvironmentClone
          0     334     334     781   Environment.OverrideEnvironment
          0     913     913    1546   Executor.Executor
          0       0       0    1712   Executor.Null
          0     980     980    2987   Node.FS.Base
          0      93      93     726   Node.FS.Dir
          0     293     293    1667   Node.FS.File
          0    1277    1277    3284   Node.Node
```

So about a 50% reduction

pd0wm, 2021-11-23:
When I build without the PR it's pretty similar for some reason. Not sure how accurate this memory debugging tool is:

```
Memory before reading SConscript files:     24268800
Memory after reading SConscript files:     351797248
Memory before building targets:            756047872
Memory after building targets:             810848256
Object counts:
       pre-   post-    pre-   post-   
       read    read   build   build   Class
          0     659     659    2908   Action.CommandAction
          0      20      20      20   Action.CommandGeneratorAction
          0     101     101     101   Action.FunctionAction
          0      78      78      78   Action.LazyAction
          0      15      15      15   Action.ListAction
          0     208     208     208   Builder.BuilderBase
          0      18      18      18   Builder.CompositeBuilder
          0     555     555     555   Builder.OverrideWarner
          0       5       5       5   Environment.Base
          0      10      10      10   Environment.EnvironmentClone
          0     342     342     795   Environment.OverrideEnvironment
          0     941     941    1574   Executor.Executor
          0       0       0    1717   Executor.Null
          0    1007    1007    3014   Node.FS.Base
          0      97      97     730   Node.FS.Dir
          0     311     311    1685   Node.FS.File
          0    1312    1312    3319   Node.Node
```

PR #23011 (2021-11-23): don't build internal Kalman objects on device
closes: #23006
closed 2021-11-23 by pd0wm