commaai/openpilot #23026: Security Vulnerabilities Found within OpenPilot Source Code

chadknowlton, 2021-11-24:
My team and I have been reviewing OpenPilot as part of our Semester project at University and part of this review was running an automated scanning tool on OpenPilot's source code. It's conclusions can be found here:

https://sonarcloud.io/summary/overall?id=Rafterman29_openpilot

We have noted several key vulnerabilities including CWE-200: Exposure of Sensitive Information to an Unauthorized Actor, CWE-326: Inadequate Encryption Strength, CWE-259: Use of Hard-coded Password, CWE-327: Use of a Broken or Risky Cryptographic Algorithm, CWE-401: Missing Release of Memory after Effective Lifetime, and more as listed in the SonarCloud scan linked above. Anyone is free to email me at chadknowlton@unomaha.edu for any more specific questions or direct access to my team's private repository in regards to our software review for this course.

chadknowlton, 2021-11-24:
For further review of the project we have also created Design Documentation, Assurance Cases, Software Security Engineering diagrams (including Misuse cases), and more documentation that can all be found within our now public repository at: https://github.com/chadknowlton/PrivateOpenPilotRepo

Again, feel free to reach out to me at chadknowlton@unomaha.edu if there are any questions. I wanted to supply our entire repository in case it is of use for anyone working on these security enhancements in the future.

pd0wm, 2021-11-25:
Quickly looked over that scan result, but didn't see anything concerning. Do you think there is anything in those results that is actually relevant? Feel free to make the rest of your report public.

Between all the false positives it found two memcpy issues in boardd. One of them is already fixed.
 - (fixed) https://sonarcloud.io/project/issues?id=Rafterman29_openpilot&open=AX0_THL8gvHzTIyGNIWv&resolved=false&types=BUG
- @briskspirit Is this one still relevant? https://sonarcloud.io/project/issues?id=Rafterman29_openpilot&open=AX0_THL8gvHzTIyGNIWw&resolved=false&types=BUG

chadknowlton, 2021-11-25:
Our main concerns are listed in our CodeAnalysis.md document: 
https://github.com/chadknowlton/PrivateOpenPilotRepo/blob/main/CodeAnalysis.md

But all of the issues we found (that aren't false positives) seem to be relatively quick changes. For instance in athnad.py there is the following code snippit:
'upload_id = hashlib.sha1(str(item).encode()).hexdigest()'

SHA-1 is depricated and is no longer secure so it could be quickly replaced with SHA-256 to have a more secure hashing algorithm.

You are correct in saying that, for the most part, there isn't much that is overly concerning. We mostly wanted to bring up our results due to the quick-fix nature of any of the issues that we did find. However, whenever you have the time feel free to look at the document I linked and let me know what you think.

briskspirit, 2021-11-25:
> * @briskspirit Is this one still relevant? https://sonarcloud.io/project/issues?id=Rafterman29_openpilot&open=AX0_THL8gvHzTIyGNIWw&resolved=false&types=BUG

No, tail_size is never will be bigger than tail buffer.

pd0wm, 2021-11-26:
@briskspirit I assume it can happen if the USB packet is malformed, which is why the scanner triggered? Maybe we should put an assert before the memcpy so it just crashes instead of silently memcpying outside the buffer.

briskspirit, 2021-11-26:
@pd0wm it is also not really possible as it depends on max USB frame size of 64 bytes, but I will add assertion to be safe.

closed 2021-11-29 by pd0wm