commaai/openpilot #23056: Audio cutoff during long turn-exceeded alerts

jyoung8607, 2021-11-29:
### Describe the bug

A few times during our holiday weekend travels, I noticed the audio alert for steerSaturated can be abruptly truncated in the middle of the file, if the alert happens to resolve while audio is being played. I can't recall hearing this before in multiple years of openpilot use.

I'm not sure if you consider this a bug, but I found it pretty jarring/janky. I also think there were some occasional snap/crackle artifacts right at the end as playback cuts off (maybe amp shutoff?).

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

3cfdec54aa035f3f|2021-11-27--22-06-35

### openpilot version

master @ b51da61dc4401efadfe15b90ce65ab24c80cf9e0 + MQB long development stuff

### Additional info

I couldn't write down the exact timestamp while driving, but I think one of the steerSaturated events in the 10:57-10:58pm range or possibly the one at 11:10pm may demonstrate the problem.

adeebshihadeh, 2021-11-29:
We're missing `libqt5multimedia5-plugins`, which will be [included in AGNOS 3](https://github.com/commaai/agnos-builder/commit/1fef610c7f80839100b42968ff13532f94080e88). Hopefully, using the proper audio plugin will fix this.

adeebshihadeh, 2021-11-30:
@deanlee want to check this out? May want to see if installing the proper plugins fixes it: `sudo apt-get update && sudo apt-get install libqt5multimedia5-plugins`. Not sure how we're playing sound right now with no audio backend.

deanlee, 2021-11-30:
I am doing testing.

adeebshihadeh, 2021-12-03:
Should be fixed by #23076

adeebshihadeh, 2021-12-11:
Reverting #23076 and reopening this. The overlapping sounds it can cause are confusing, and I don't think the artifacts are from stopping looping sounds. We're still hearing reports of rare sound artifacts on non-looping sounds like engage and disengage.

deanlee, 2021-12-12:
multiple QSoundEffect will create multiple audio outputs. It is not clear if it is related to this issue
[image]

adeebshihadeh, 2021-12-12:
I believe that depends on the QAudioDevice used. Right now we let it use the default, but that's likely different on the comma three than the PC.

adeebshihadeh, 2021-12-15:
The volume changing during playback is a known issue. Usually the effect is fairly minor, but perhaps this is the cause of all the weirdness we're seeing. I think fixing the volume setting is a good place to start @deanlee.

adeebshihadeh, 2022-06-02:
I haven't hit this in months. @jyoung8607 is this still an issue for you?

jyoung8607, 2022-06-02:
I don't think so.

closed 2022-06-02 by jyoung8607