commaai/openpilot #23064: Ensure git repo isn't touched outside overlay

adeebshihadeh, 2021-11-29:


pd0wm, 2021-12-01:
I bet we can replicate some of the git branch/commit functions in python by reading the git config directly. This way we can ensure it does't write anything.

jyoung8607, 2021-12-01:
Back when I wrote the updater, I wasn't aware of this much faster and cheaper way to check for remote changes:

[image]

I'm assuming `git rev-parse HEAD` is a readonly op, I wouldn't think it would try to check/refresh things like `git status` and friends, but I haven't checked. If it does make changes, the local commit could be checked just once at manager startup. `git ls-remote [remote [branch]]` shouldn't be making any local changes.

Using this method, everything on C2/C3 would stay readonly when just checking if an update exists. We wouldn't even need to set up the overlay, we'd only need to do it when an actual update becomes available. Ditching the overlay setup and the full `git fetch` operation every ten minutes seems like a good resource savings, especially on C2.

Extra credit: the git repo protocol is well-documented, and `updated` could be taught to call `ls-refs` methods against GitHub directly without launching an external process at all.

adeebshihadeh, 2022-09-15:
Haven't seen any issues on agnos so far. @jyoung8607 #25724 uses `git ls-remote`

closed 2022-09-15 by adeebshihadeh