commaai/openpilot #23087: honda passport 2021 controls mismatch

justinmccaffrey, 2021-12-01:
### Describe the bug

I added support for my passport with this PR https://github.com/commaai/openpilot/pull/22781  
Updated to 8.11 and on activating openpilot i got a lot of steering torque warnings for a couple seconds, then a control mismatch with immediate disable.  
I'm not sure where i messed the car model up, i believe it should behave exactly like the 2019 pilot (the model it was using before)

### What hardware does this issue affect?

comma three

### Which car does this affect?

Honda Passport 2021

### Provide a route where the issue occurs

684e8f96bd491a0e|2021-11-30--17-59-26

### openpilot version

0.8.11

justinmccaffrey, 2021-12-01:
somehow useradmin has it ID'd as `HONDA PASSPORT 2021` but under LiveParameters it fingerprints as `HONDA PILOT 2019`
```
                                     {'key': 'LiveParameters',
                                      'value': b'{"carFingerprint": "HONDA PILOT 2019", "steerRatio": 17.454078674316406, "stiffnessFactor": 0.999989'
                                               b'3307685852, "angleOffsetAverageDeg": 0.7715247869491577}'},
```

adeebshihadeh, 2021-12-01:
This was likely due to https://github.com/commaai/panda/pull/789. Can you test #23089? You can setup that branch by uninstalling and using this URL: `installer.comma.ai/commaai/fix-ridgeline`.

Verylukyguy, 2021-12-01:
I have had a similar issue with my vehicle since the new release of .0.8.11.

I have a GMC Acadia and on a few occasions my Comma Two has loaded controls for the new Cadillac Escalade ESV 2016 that was recently added.

See routes:
e38a02cae1ef1e0c|2021-11-27--19-37-31
e38a02cae1ef1e0c|2021-11-30--16-49-43

justinmccaffrey, 2021-12-01:
> This was likely due to [commaai/panda#789](https://github.com/commaai/panda/pull/789). Can you test #23089? You can setup that branch by uninstalling and using this URL: `installer.comma.ai/commaai/hotfix-ridgeline`.

Yes that branch works like normal.  Though `AudibleAlert.chimeWarningRepeat` still haunts my dreams...  
Hope we can get that merged, thanks

sailingman40, 2021-12-01:
Same issue for me, updated to .8.11 yesterday, Honda Ridgeline, seconds after set, controls mis-match error..
Was fine on prior release... something be broke on this release that needs to be backed out.

adeebshihadeh, 2021-12-01:
@Verylukyguy this is a Honda specific bug

adeebshihadeh, 2021-12-02:
Thanks for reporting this bug. The change has been reverted in release: https://github.com/commaai/openpilot/pull/23090. A new test has been added to catch this class of bugs in #23093. #23099 is a PR to fix the actual issue. The test says it's good, but feel free to double check on your car.

closed 2021-12-02 by adeebshihadeh

arafatmamun, 2021-12-02:
I have the same issue with Honda pilot 2019 after updating .0.8.11

adeebshihadeh, 2021-12-02:
Go to settings -> software then click check for update to pull the fix.

sailingman40, 2021-12-05:
New 8.11 release installed today.. solved the mis-match error on my ridgeline.. thank you and good to have automation test.