commaai/openpilot #23114: Can not start tool juggler on freh install

schenlap, 2021-12-03:
### Describe the bug

Following
https://web.blogcopas.eu.org/host-https-github.com/commaai/openpilot/blob/master/tools/README.md
and
https://web.blogcopas.eu.org/host-https-github.com/commaai/openpilot/tree/master/tools/plotjuggler
for installation gives:
```
((openpilot) ) :/foobar/openpilot/tools/plotjuggler# ./juggle.py -h
Traceback (most recent call last):
  File "./juggle.py", line 9, in <module>
    from common.basedir import BASEDIR
ModuleNotFoundError: No module named 'common'
((openpilot) ) :/foobar/openpilot/tools/plotjuggler# 
```
Something is missing in the readme

### OS Version

Ubuntu 21.10

pd0wm, 2021-12-03:
Your `PYTHONPATH` environment variable is not pointing to your openpilot installation.

schenlap, 2021-12-04:
Openpilot assumes it is cloned into HOME and sets PYTHONPATH to $HOME/openpilot.

closed 2021-12-04 by schenlap