commaai/openpilot #23124: Disengagement via braking sometimes triggers unnecessary red alert

burghoff, 2021-12-04:
### Describe the bug

Since updating to 0.8.11, a frequent occurrence has been that disengaging via braking causes an immediate red alert.

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

930ad2d3aa9a7438|2021-12-04--10-08-21

### openpilot version

0.8.11

ffhspa, 2021-12-05:
If its the cruiseDisabled Event Alert, this happens on Honda Civic Bosch Diesel as well. If you gently press brake  cruiseDisabled Event gets triggered. Havent noticed this on OP 0.8.10 only on 0.8.11

burghoff, 2021-12-05:
Yes, the message is that cruise is disabled so I assume that is the cruiseDisabled Event.

ffhspa, 2021-12-16:
Any update on this Issue? Not knowing if pressing Brake will trigger Red Alert is annoying.  Please fix this in 0.8.13 or bring a Hotfix in 0.8.12

burghoff, 2021-12-17:
Just confirmed that the issue persists in 0.8.12.

ffhspa, 2021-12-17:
I can also confirm that this issue  exists in 0.8.12 
@adeebshihadeh

cfranhonda, 2021-12-17:
Same experience. It’s when you barely tap the brakes and continue coasting. When you brake and continue braking it’s fine.

ffhspa, 2022-01-03:
Dear Comma Team
any updates on this issue? 
Would be great if it could be solved with OP 0.8.13

ffhspa, 2022-01-12:
Any Update on this issue? @pd0wm  Thanks in advance,

closed 2022-02-10 by adeebshihadeh