commaai/openpilot #23127: Comma 3 not shutting down after car is turned off

schuette13, 2021-12-05:
### Describe the bug

My comma 3 is not properly shutting down after my 2019 Subaru Forester is turned off and the driver exits the vehicle.
It continuously uploads video to connect.comma.ai to the point that Comma banned my device.  It has been reinstated now, and I make sure to unplug the comma 3 every time I leave the car.
Because it is not shutting down it also seems to ignore/bypass the code that is supposed to shut it off if my car's battery voltage becomes too low.  This has caused my car's battery to be fully drained on a couple of occasions after a few day of not driving the car and not unplugging the comma 3.

There is a video attached that starts at step 3 below.  Ways to reproduce the video:
1. Turn on the car and the comma
2. Turn off the car and exit the vehicle
3. The comma's screen will go black, but then after a few seconds will start displaying the "landing" page.
4. The screen will briefly show what the camera sees and then it will show the landing page again.
5. Wait long enough and the car battery is dead and the comma has uploaded a large amount of video files. 

Before flashing agnos, it would show the landing page, then show what the camera sees with the words "Openpilot unavailable waiting for controls to start."

Things I've tried:
1. Disconnect and reconnect all cables
2. Uninstall/reinstall Openpilot
3. Flash agnos
4. Ask the Subaru discord channel if anyone has seen this (no one had)
5. Contact support which asked that I file this bug report.

https://user-images.githubusercontent.com/7681453/144728489-5f9f0944-c984-4617-86c3-882c00578665.mp4

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

cb21d4ecba09f745|2021-12-03--17-02-52

### openpilot version

0.8.11

adeebshihadeh, 2021-12-05:
I'm seeing lots of short routes from your device starting on 12/1, which I'm assuming is when your device was unbanned. This means the ignition signal is going high very briefly, but by the time the logger starts up, ignition is low again on all the routes I checked. When did this start happening? It's possible that the CAN ignition added in 0.8.11 is causing this, but I think it's unlikely since it seems your device was initially banned in September, when I'm assuming this issue started.

schuette13, 2021-12-05:
To my knowledge it's been happening the whole time I've owned the device.
I've never owned another comma device so I'm not positive what the behavior
is supposed to be.
Back in September I noticed some longer drives from the garage but thought
that was normal because I didn't know any better. Then my device stopped
uploading drives (banned) and it killed my battery after not driving for a
couple days. I've been very careful with unplugging it ever since.

My original email to comma support was on both version 0.8.8 and 0.8.9

Let me know if I can give any additional information.

Michael

adeebshihadeh, 2021-12-05:
I checked again, and looks like you're right. It was just much less frequent when you first got your device. This is likely a hardware issue. I'll let the support team know, and we'll get your hardware swapped out.

closed 2021-12-05 by adeebshihadeh

RobustAlpaca, 2022-01-27:
I am having this exact same issue with my C3 and 2020 Hyundai Kona. Need help, please.

schuette13, 2022-01-27:
> I am having this exact same issue with my C3 and 2020 Hyundai Kona. Need help, please.

I still need to reach out Comma to get a new unit. I'm still using my original device.

RobustAlpaca, 2022-01-27:
Did you determine it was the C3 itself or the harness box?