commaai/openpilot #23144: Improve comma three thermals

pd0wm, 2021-12-06:
- [ ] Switch to powersave governor when offroad
- [ ] Look into Quectel power savings
- [ ] Lower fan control setpoint

adeebshihadeh, 2022-04-27:
I think we should also re-evaulate the threshold to go onroad. Currently it's set at 79.5C, but a device that goes onroad at that temperature will either narrowly avoid going into the red band or immediately jump into it. We'll either need to keep the device cooler while it's offroad or build a better experience when we don't immediately go onroad. 

[image]

jyoung8607, 2022-06-02:
`3cfdec54aa035f3f|2022-05-29--16-38-08` is an example of the problem Adeeb described, a startup from *just* below the offroad danger temp that goes *just* into the red a short time after, resulting in a very long delay before openpilot was engageable.

Hubblesphere, 2022-07-22:
Lapse in overheat protection. Off road overheated able to engage temporarily when going onroad before detecting an overheated condition again.

Route:  8d8427a2116eace2|2022-07-22--13-00-54--0

jyoung8607, 2022-08-24:
Another example: `3cfdec54aa035f3f|2022-08-24--15-21-31`

goodchoom, 2023-04-20:
When the temperature outside is 70°F or hotter and my car has been sitting outside for an extended period of time, I always remote start my car with the A/C set to LO several minutes in advance so by the time I enter my car it's comfortable for me to drive. Despite the cabin having cooled down, I am sometimes unable to engage openpilot because it went onroad just below the offroad danger temp and then overheated while the cabin was hot. When this happens I have to wait/drive manually for several more minutes until the device cools down enough to be able to engage openpilot again.

Here is a route of this issue occuring: 4f3e4f2f6d94e940|2023-04-20--12-27-08--0

adeebshihadeh, 2023-08-07:
> When the temperature outside is 70°F or hotter and my car has been sitting outside for an extended period of time, I always remote start my car with the A/C set to LO several minutes in advance so [...]

Recent releases have addressed the onroad overheating. Let us know if you still have issues.

closed 2023-08-07 by adeebshihadeh