commaai/openpilot #23152: start the simulation but nothing happend.

Torch-HXM, 2021-12-07:
### Describe the bug

(1) I did all in openpilot/tools/README.md
(2) Then,I run ./start_carla.sh in one terminal and ./start_openpilot_docker.sh in another.
(3) There just come out a ui and nothing happend.
(4)Here is the picture.
[image]

### OS Version

ubuntu20.04

### openpilot version or commit

0.8.12

### Additional info

(1)./start_carla.sh
[image]
(2)./start_openpilot_docker.sh
(2.1)./bridge.py
[image]
(2.2)./launch_openpilot.sh
[image]

pd0wm, 2021-12-07:
Can you post the full output of 2.2? There must be a reason why it's not going onroad. Just tried reproducing on latest master, but works fine for me.

Torch-HXM, 2021-12-08:
> 

thank you for your reply.here is the full output of 2.1 and 2,2
(2.1)
[image]
I did as it says:add the variable ahead of ./bridge and the output is as follows.
[image]
(2.2)
[image]

Torch-HXM, 2021-12-08:
> Can you post the full output of 2.2? There must be a reason why it's not going onroad. Just tried reproducing on latest master, but works fine for me.

thank you for your replay.Here is the full output of 2.1 and 2.2.
(2.1) ./bridge
[image]
I did as it says:add the variable ahead of ./bridge and the output is as follows.
[image]
(2.2) ./launch_openpilot.sh
[image]

pd0wm, 2021-12-08:
Don't see a reason why it wouldn't go onroad. You can print `startup_conditions` in thermal (https://github.com/commaai/openpilot/blob/master/selfdrive/thermald/thermald.py#L352) to debug this further.

pd0wm, 2021-12-14:
Closing due to inactivity. I recommend the #openpilot-simulation channel on discord.comma.ai, that's probably a better place for these debugging questions.

closed 2021-12-14 by pd0wm