commaai/openpilot #23162: reboot: low on memory

tecandrew, 2021-12-07:
### Describe the bug

while using OP it gave the loud alert to take control then prompted to reboot the device because it was low on memory. initially happened towards the end of a 2 hr drive.

### misc:
- just obtained the device not too long ago
- was actively routing to destination using nav
- around a 2 hr drive

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

eb5a50f9e98f50b9|2021-12-06--16-44-30 

### openpilot version

0.8.11-release

### Additional info

another user on discord `#nav-feedback` said it also happened to them recently (their drive: `66253de0fd139602|2021-12-03--12-47-46--35`)

happened again earlier today on a shorter drive: towards the end of `eb5a50f9e98f50b9|2021-12-07--12-32-36--6`

redacid95, 2021-12-08:
This is a known issue and will be fixed in 0.8.12 release. Master currently has the fix in place based on what I have seen in their milestone tracker. https://github.com/commaai/openpilot/issues/22298

pd0wm, 2021-12-08:
@redacid95 the fix should have already made it into the 0.8.11 release. This might be a new issue. I'm looking into it.

closed 2021-12-08 by pd0wm

ananyaget, 2022-04-21:
Currently having this exact problem C3 sunnypilot Hyundai ioniq phev 2022.
About 1:45 min into drive low memory reboot issue.

ananyaget, 2022-04-21:
On 0.8.12 and low memory issue is still happening