commaai/openpilot #23163: Add "Flash Panda" button in Settings

BadOlives, 2021-12-08:
Pretty self-explanatory. Add a button in the Settings menu that a user can press to flash panda without having to SSH into the device and run the commands located in Flashing Panda wiki.

adeebshihadeh, 2021-12-08:
Panda flashing is handled automatically by pandad.

closed 2021-12-08 by adeebshihadeh

BadOlives, 2021-12-08:
> Panda flashing is handled automatically by pandad.

Panda does not always flash successfully, causing OP to not function properly when engaged.

jyoung8607, 2021-12-08:
> Panda does not always flash successfully, causing OP to not work at all or not function properly when engaged (if you are able to)

Signature of Panda's current image is compared with what openpilot wants at every startup, and openpilot will automatically reflash Panda if it doesn't match. In case of failure, openpilot won't drive the car at all with mismatching Panda FW.

Failures should be rather rare, and generally resolved by restarting the device. It's doing mostly the same thing you'd do from the command line, so if the command line was going to work for you, pandad would have already taken care of it.

To the extent you run into folks saying "it finally worked after I reflashed Panda 7 times", reflashing Panda generally isn't the thing that actually fixed whatever problem they were having.

BadOlives, 2021-12-08:
> Signature of Panda's current image is compared with what openpilot wants at every startup, and openpilot will automatically reflash Panda if it doesn't match. In case of failure, openpilot won't [...]

Typically, I'll reflash the panda only once and everything starts working again. The issue I had was my steering wheel was being controlled by OP, however, it wasn't full steering like it normally does, causing OP to not center me in my lane quick enough and having to take manual control. After reflashing the panda, the issue was fixed.

jyoung8607, 2021-12-08:
> Typically, I'll reflash the panda only once and everything starts working again. The issue I had was my steering wheel was being controlled by OP, however, it wasn't full steering like it normally [...]

I don't wish to be unkind, but this 100.00% an example of the placebo Panda flashing I mentioned.