commaai/openpilot #23166: Could not find install_carla.sh as manual described in “https://github.com/commaai/openpilot/wiki/CARLA”

skyw0889, 2021-12-08:
### Describe the bug

During trying simulate openpilot in PC,  during environment prepare phase could not find install_carla.sh as manual described in “https://github.com/commaai/openpilot/wiki/CARLA”, the whole description section was following:
"
**Install openpilot and carla**
sudo apt install git

git clone https://github.com/commaai/openpilot.git

cd openpilot/tools/sim

./install_carla.sh
"
So if any info about this issue would be appreciated, thanks a lot!

### OS Version

Ubuntu20.04 LTS 

### Additional info

Manual link:
https://github.com/commaai/openpilot/wiki/CARLA

pd0wm, 2021-12-08:
The wiki page is outdated. The updated instructions are here: https://github.com/commaai/openpilot/blob/master/tools/sim/README.md

I'll update the wiki to point to the readme

closed 2021-12-08 by pd0wm

skyw0889, 2021-12-08:
OK， Great， very appreciate for your feedback!  : )

skyw0889, 2021-12-08:
> OK， Great， very appreciate for your feedback! :