commaai/openpilot #23180: Every time simulate on PC run "sudo ./start_openpilot_docker.sh" would redownload lot's of packages

skyw0889, 2021-12-09:
### Describe the bug

During trying simulate on PC, i found every time simulate on PC run "sudo ./start_openpilot_docker.sh" would redownload lot's of packages, is this necessary or could reuse pre-download data(if i am wrong please point out), so if any information about this issue would be appreciated, thanks a lot!

### OS Version

Ubuntu20.04 LTS 

### Additional info

Command: sudo ./start_openpilot_docker.sh

Manual:
https://gitee.com/skywll/openpilot/wikis/CARLA

pd0wm, 2021-12-09:
It downloads the latest openpilot docker container. You can comment out the docker pull if you want to keep using the same version: https://github.com/commaai/openpilot/blob/master/tools/sim/start_openpilot_docker.sh#L9

closed 2021-12-09 by pd0wm

skyw0889, 2021-12-09:
Great, i would take a try, very appreciate your feedback, also a very quickly feedback!!!