commaai/openpilot #23188: LogReader unable to read some rlog.bz2 files.

uyrusali, 2021-12-10:
### Describe the bug

[rlog.zip](https://github.com/commaai/openpilot/files/7691221/rlog.zip)

``` python
#%%
import os
import pandas as pd
# from tools.lib.logreader import LogReader
from tools.lib.robust_logreader import RobustLogReader
from datetime import datetime
import glob
import pytz
from pathlib import Path
import tools.lib.logreader
parentDir="/home/ubuntu/home/ubuntu/comma_logs/20211209_bump/2021-12-09--11-39-41--0/"
files=[]
keys=["latitude","longitude","altitude","speed","timestamp"]

csv_dir=os.path.join(parentDir,"comma_csv")

files.append('rlog.bz2')

for file in files:
    print(file)
    # csv_file=os.path.join(csv_dir,'{}.csv'.format(file.split("/")[10]))
    csv_file = '1.csv'
    lr=RobustLogReader(file)
    logs=list(lr)
    print([set(l.which() for l in logs)])
    gps_location_external=list(set([l.gpsLocationExternal for l in logs if l.which() == 'gpsLocationExternal']))
    sensorEvents = print(list(set([l.sensorEvents for l in logs if l.which() == 'sensorEvents'])))
```
Unable to read attached rlog file using the script above. I was able to read other rlog files, and qlog files from this trip using the same script, but reading rlog files from this trip gives the following error. Tried both LogReader and RobustLogReader.

```
Traceback (most recent call last):
  File "capnp/lib/capnp.pyx", line 1165, in capnp.lib.capnp._DynamicStructReader._which
RuntimeError: Member was null.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/PycharmProjects/IMU/micgihan_converter.py", line 32, in <module>
    print([set(l.which() for l in logs)])
  File "/home/ubuntu/PycharmProjects/IMU/micgihan_converter.py", line 32, in <genexpr>
    print([set(l.which() for l in logs)])
  File "capnp/lib/capnp.pyx", line 1182, in capnp.lib.capnp._DynamicStructReader.which.__get__
  File "capnp/lib/capnp.pyx", line 1168, in capnp.lib.capnp._DynamicStructReader._which
capnp.lib.capnp.KjException: Attempted to call which on a non-union type
```

### OS Version

Ubuntu 20.04

pd0wm, 2021-12-10:
Is your `cereal` repo (specifically log.capnp), on the same version as was used to generate the logs?

uyrusali, 2021-12-10:
That's true, it was my mistake. The problem was using the older cereal repo in a virtual environment.

Problem solved with using the most recent cereal version.

Thank you @pd0wm !

closed 2021-12-10 by uyrusali