commaai/openpilot #23194: Poor GPS Reception (rare)

ErichMoraga, 2021-12-11:
### Describe the bug

After driving a couple miles, I noticed "Poor GPS reception" persist on the UI, as seen in...
https://github.com/commaai/openpilot/issues/22687
https://github.com/commaai/openpilot/issues/20731

[image]

I thought this was fixed...
https://github.com/commaai/openpilot/pull/22712

Perhaps it's not totally resolved yet.

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

b29a96b5483f601c|2021-12-11--10-00-37

### openpilot version

0.8.11

### Additional info

Over the last dozen or so hours I've driven on this branch (which is very close to stock), I only recall seeing this error ~3 or 4 times.

adeebshihadeh, 2021-12-11:
@robbederks can you give this a quick look next week to see if it's quick fix for 0.8.12?

robbederks, 2021-12-11:
Sure, I'll take a look!

closed 2021-12-13 by robbederks

ErichMoraga, 2021-12-13:
> The GPS in your device seems to be degraded in general compared to the fleet, which probably is the main reason for the slow-ish time-to-fix. As a comparison, this is a plot of the accuracy distribution from all your drives on this device, while the fleet seems to average more around sub 1m accuracy. Are you still running your EPHOT device?
> 
> Also, the issue/PR you linked are only relevant to C2 :)
> 
> [image]

I thought the C3 used the same UBlox M8 module as the C2...
https://discord.com/channels/469524606043160576/871838269405556736/919641352571060295

Anyway, yes, I'm still using my EPHOT RevG.  Comma wanted me to exchange it anyway, so I'll reach out to @VirtuallyChris to start working on that.

robbederks, 2021-12-14:
> > The GPS in your device seems to be degraded in general compared to the fleet, which probably is the main reason for the slow-ish time-to-fix. As a comparison, this is a plot of the accuracy distribution from all your drives on this device, while the fleet seems to average more around sub 1m accuracy. Are you still running your EPHOT device?
> > Also, the issue/PR you linked are only relevant to C2 :)
> > [image]
> 
> I thought the current C3 used the same UBlox M8 module as the C2... https://discord.com/channels/469524606043160576/871838269405556736/919641352571060295
> 
> Although I realize they're on different OSes, the bug was with the `boardd` and UBlox interaction, which should still be shared b/w C2 & C3, right?

That's all true, but the main difference is that on the C2 the ublox is connected through the panda, while on the C3 the ublox is connected straight to the SOM. The bug was only relevant to C2 since there it would turn off the ublox when the panda would go into low power mode, which is not the case on C3 :)

> Anyway, yes, I'm still using my EPHOT RevG. Comma wanted me to exchange it anyway, so I'll reach out to @VirtuallyChris to start working on that.

Ok, great! The rev Gs don't have all the RF mitigations we have added to the next revisions, so that explains the less than desirable performance!

ErichMoraga, 2021-12-14:
Thanks for the explanation.  Seems this issue is getting increasingly more common on my C3, since I just hit it again today...
`b29a96b5483f601c|2021-12-14--10-48-11`

ErichMoraga, 2021-12-21:
@robbederks I received a new, production C3 today, and it got `Poor GPS reception` as segment 7 started...
`e6448e8ba54a1aab|2021-12-20--21-26-43`

```
'params': {'entries': [{'key': 'ApiCache_Device',
 'value': b'{"alias": "C3 e6448e8ba54a1aab", "athena_host": "prod-comma-public-athena-0.prod-comma-public-athena'
          b'.production.svc.cluster.local", "device_type": "three", "dongle_id": "e6448e8ba54a1aab", "ignore_upl'
          b'oads": null, "last_athena_ping": 1640034870, "last_gps_accuracy": null, "last_gps_bearing": null, "l'
          b'ast_gps_lat": null, "last_gps_lng": null, "last_gps_speed": null, "last_gps_time": null, "public_key'
          b'": "-----BEGIN PUBLIC KEY-----...-----END PUBLIC KEY-----\\n", "serial": "a2'
          b'25f199", "sim_id": "8901260195705794885", "trial_claimed": true, "is_paired": true, "is_owner": fals'
          b'e, "prime": true, "openpilot_version": "0.8.12"}'},

'gpsOK': False,
'gpsTimeOfWeek': 0.0,
'gpsWeek': 0,

'logMonoTime': 475996471432,
 'valid': True}
{'controlsState': {'aEgoDEPRECATED': 0.0,
                   'aTarget': 0.0,
                   'aTargetMaxDEPRECATED': 0.0,
                   'aTargetMinDEPRECATED': 0.0,
                   'active': False,
                   'alertBlinkingRate': 0.0,
                   'alertSize': 'mid',
                   'alertSound': 'none',
                   'alertStatus': 'normal',
                   'alertText1': 'Poor GPS reception',
                   'alertText2': 'If sky is visible, contact support',
                   'alertType': 'noGps/permanent',
                   'angleModelBiasDEPRECATED': 0.0,
                   'angleSteersDEPRECATED': 0.0,
                   'awarenessStatusDEPRECATED': 0.0,
                   'canErrorCounter': 0,
                   'canMonoTimeDEPRECATED': 0,
                   'canMonoTimes': [],
                   'cumLagMs': -12.046008110046387,
                   'curvature': -0.05189300701022148,
                   'decelForModelDEPRECATED': False,
                   'decelForTurnDEPRECATED': False,
                   'driverMonitoringOnDEPRECATED': False,
                   'enabled': False,
                   'engageable': True,
                   'forceDecel': False,
                   'gpsPlannerActiveDEPRECATED': False,```

robbederks, 2021-12-21:
Weird, just looked at the rlogs, and there's no signal _at all_ for the whole drive. 
Also only see 1 route for that device so far, will look again in a few days to see if this is persistent or a one-off bug.

ErichMoraga, 2021-12-21:
> Weird, just looked at the rlogs, and there's no signal _at all_ for the whole drive. Also only see 1 route for that device so far, will look again in a few days to see if this is persistent or a one-off bug.

Yeah, I noticed that too.  When I initially got it, loaded software via 12V connection, and activated Prime, it seemed to know where it was generally located (per Comma Connect).  I'll drive it a few more times (maybe 1 or more per day).  Not stressed or anything.  Thanks for looking, and enjoy the holidays :partying_face:

robbederks, 2021-12-21:
Yeah, iirc connect uses other data like the wifi SSIDs in range to do rough positioning. Hopefully this isn't another HW issue...
Feel free to DM me when you have another route!