commaai/openpilot #23224: [$200 Bounty] Toyota: cancel cruise without chime

adeebshihadeh, 2021-12-15:
The way openpilot cancels cruise control on most Toyotas has the side effect of a chime from the car and a momentary fault on the dash. When openpilot sends the cancellation message, it also sends a message to play a good chime to mask the fault chime that would play, so start by [removing that](https://github.com/commaai/openpilot/blob/fa62b9d3f9882ec33559ceb7377fa991be1d4fb5/selfdrive/car/toyota/toyotacan.py#L82).

The Lexus IS uses a [different cancellation message](https://github.com/commaai/openpilot/blob/fa62b9d3f9882ec33559ceb7377fa991be1d4fb5/selfdrive/car/toyota/carcontroller.py#L94) than the rest of the Toyotas, so that's probably a good place to start.

$200 for a merged PR that meets this criteria:
* no chime sound or even momentary fault display
* reliable, i.e. always immediately cancels cruise
* generic across all Toyotas

Feel free to open a PR once #1 and #2 are done to lock the bounty pending confirmation of #3.

cydia2020, 2021-12-15:
> The Lexus IS uses a different cancellation message than the rest of the Toyotas, so that's probably a good place to start.

Tried that already if anyone wants to work on this, doesn't cancel cruise on the Prius.

cydia2020, 2021-12-15:
I will also be documenting some findings here, cruise button states (at least on the Prius, are available over CAN, however, they are not available on the ADAS CAN, they are only on the yaw sensor / air bag / hybrid control CAN as far as I can tell.

```
BO_ 865 PCM_BUTTONS : 7 PCM
 SG_ RES_BTN : 5|1@0+ (1,0) [0|0] "" XXX
 SG_ SET_BTN : 4|1@0+ (1,0) [0|0] "" XXX
 SG_ CANCEL_BTN : 3|1@0+ (1,0) [0|0] "" XXX
 SG_ MAIN_ON : 2|1@0+ (1,0) [0|0] "" XXX
```

sshane, 2021-12-16:
Same findings here for TSS2 Camry, I tried sending different variations of `PCM_CRUISE`, `PCM_CRUISE_2`, `PCM_CRUISE_SM`, `DSU_CRUISE` with some setting the cancel button and others setting `MAIN_ON` to 0. Didn't see any effects. Also explored bit 29 on `ACC_CONTROL`  and bit 3 on `PCM_CRUISE_2` which always seems to be flipped compared to `CRUISE_ACTIVE`, but cruise remains active.

nelsonjchen, 2021-12-17:
Some thoughts, scattered like everyone else's. Unfortunately all I have is speculation.

* @AlexandreSato and friends' quest to get full speed radar radar on their Corollas was kind of interesting. It involved sending some sort of wakeup to the car before the car starts. The current thread participants' vehicles all have full speed radar cruise control so maybe we can't see some form of cancel that these out-of-the-box non-full-speed vehicles use or send that we could also use as well. I'm speculating that we might be able to use the same command that these users do to enable full speed ACC but flipped in a way to "disable" full speed radar cruise control on our own vehicles so maybe we can observe that particular cancellation's message. This might be a too complicated red herring and the cancellation may not even come from the camera.
  * On that note, unfortunately, @AlexandreSato's C2's Leeco died and it was replaced with a Chinese clone that is presumably talking to Retropilot APIs. tape's should still be talking to comma's server I think?
* How come we can spam resume?
* I wonder if the 3 minute timeout sends a cancel message that can be used. 

Also, here's what is in my car's manual about what can cause cancelation just for easy reading in this issue itself. 

[image]
[image]

cydia2020, 2021-12-17:
> How come we can spam resume?

We can't, not on Toyotas, that's why we've the pedal.

> I wonder if the 3 minute timeout sends a cancel message that can be used.

PCM shifts the car to park in that case.

We might be able to cancel it by spamming a specific bit on the pre-crash message, hopefully it won't make noises, Toyota loves noises for some reason, I'll look into that.

nelsonjchen, 2021-12-17:
The resume spamming is something I've noticed in TSS2? Or whatever mechanism keeps the car from disengagement while stopped without having to press the pedal to resume. Sorry I should have clarified that.

adeebshihadeh, 2021-12-21:
Bumping this up to $300 after seeing how much work has gone into this so far.

sshane, 2021-12-21:
> The resume spamming is something I've noticed in TSS2? Or whatever mechanism keeps the car from disengagement while stopped without having to press the pedal to resume. Sorry I should have clarified that.

As far as I know, openpilot doesn't resume spam on any Toyota. What lets the car re-accelerate is that openpilot doesn't send the standstill bit when stopped, it only requests a lot of brake to stay stopped. Stock ACC requests standstill which requires you to press the resume button to start again

alancyau, 2021-12-22:
From wocsor:

> hey guys, was working on that lil bounty today - the ACC cancel without beeps. well i got bored with it (i didn't want to ask someone to make 2 PRs bc opendbc needs to be modded) so instead i'll just throw some hints here. would comment these on the GH issue if comma hadn't banned me.
basically, you just need to look at the PCS messages: 0x283 and 0x344. yeah i know it's taboo to talk about that here, but if you look in the 2017 toyota ref dbc, you'll see a bunch of 1 bit messages that can be manipulated. a few of them do things like ABS brake hold, ABS valve / motor control, PCS auto-brake, but there's also an ACC cancel in there that just cuts ACC without displaying an error / chime. anyway, happy hacking

Smartype, 2022-01-04:
I can do that. I accidentally found a way to do this when I testing low speed AEB braking on my Camry LE.

Smartype, 2022-01-04:
I will confirm this tomorrow and submit a PR.

Smartype, 2022-01-05:
PR: https://github.com/commaai/openpilot/pull/23404

nelsonjchen, 2022-01-05:
`071ba9916a1da2fa|2022-01-04--20-12-54--5` with @Smartype's changes on a 2020 Corolla SE Hatchback. While does deactivate without a chime, mashing the brake pedal down (e.g. enough to activate brake hold) at a standstill causes a cruise control malfunction. Needs to restart car to re-enable cruise control.

cydia2020, 2022-01-05:
Does stock AEB create a cancel event? I'm wondering how that is handled, since we're sending cancellation with the PCS command I feel like it shouldn't be sent when stock AEB is active.

Smartype, 2022-01-05:
> Does stock AEB create a cancel event? I'm wondering how that is handled, since we're sending cancellation with the PCS command I feel like it shouldn't be sent when stock AEB is active.

I don't know if stock aeb sent cancel cmd. I *accidentally* found this when I try to send brake cmd to nodsu Camry under speed 20kph. 

I have never been able to get access to a real aeb brake logs, which might help me find a way to brake Camry under 20kph.

cydia2020, 2022-01-05:
> I don't know if stock aeb sent cancel cmd. I _accidentally_ found this when I try to send brake cmd to nodsu Camry under speed 20kph.
>
> I have never been able to get access to a real aeb brake logs, [...]

Sorry I should've made myself clear, in openpilot there's an event called stockAeb, does that cancel cruise?

Smartype, 2022-01-05:
In op, stockAeb cause an alert on UI. nothing more.
I have build a camry radar intercepter, which mute op output and pass the stock acc cmd for a few seconds when stock aeb is active.

Smartype, 2022-01-05:
> Sorry I should've made myself clear, in openpilot there's an event called stockAeb, does that cancel cruise?

You are right, for safety reason, we should spam this msg when AEB is not in expected state. (Which is handled in my intercepter fw)

sshane, 2022-01-12:
Unfortunetely I tried exploring the 0x283 and 0x344 messages suggested by a comment from wocsor above and nothing made the car disengage. I set all the one bit signals to one with nothing happening except my stereo muting (must be a PCS feature where it mutes your music if you have an AEB event)