commaai/openpilot #23234: model replay random failures

adeebshihadeh, 2021-12-15:
Should be easy to catch running a loop. @deanlee want to take a look at and see why this isn't fully deterministic? There's a chance it's a test bug too.

https://jenkins.comma.life/blue/organizations/jenkins/openpilot/detail/master/1233/pipeline

Logs attached from that failed jenkins run.
[log.txt](https://github.com/commaai/openpilot/files/7722250/log.txt)

deanlee, 2021-12-15:
Is it possible that it is caused by the occasional difference in the update timing of 'lateralPlan' and 'liveCalibration' in modeld's calibration_thread?
https://github.com/commaai/openpilot/blob/a11259929ba4d6f18892ae31844501d4db1eaa36/selfdrive/test/process_replay/model_replay.py#L97-L106

adeebshihadeh, 2021-12-16:
hopefully fixed by #23237

closed 2021-12-16 by adeebshihadeh