commaai/openpilot #23252: Unintentional Acceleration while engaged

KyleGreenlaw, 2021-12-17:
### Describe the bug

Was recently driving and while engaged on the highway, Openpilot started to increase set cruise speed. Disengaged and re-engaged to see if it happened again and it did. 

Restarted the comma 3 and it did not continue after restart

2021 Hyundai Sonata Hybrid, Running Master, Comma 3

route:fe80a4e1cedec853|2021-12-15--21-34-09--10

[image]

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

route:fe80a4e1cedec853|2021-12-15--21-34-09--10

### openpilot version

0.8.12

pd0wm, 2021-12-17:
Very strange. I checked the CLU11 message, which contains the button pressed state, but neither the cluster nor openpilot was sending messages to increase the set speed.

The only thing odd was just before it started happening the button state was 7, which is not one of the known button states in openpilot, so not sure what that means.

Without a good way to reproduce this might be a tricky bug. It might not even be an openpilot bug.
[image]

pd0wm, 2021-12-17:
@gregjhogan just noticed you were using openpilot longitudinal, which I didn’t notice.

Need to look into it some more!

KyleGreenlaw, 2021-12-17:
I’m out right now, unable to upload the last drive I’ve done. I think I’ve been able to duplicate the error. If you hold down the button that adjusts distance, and then engage open pilot the speed will begin to increase.

I’ll have to get on Wi-Fi, upload it and take a look

KyleGreenlaw, 2021-12-28:
fe80a4e1cedec853|2021-12-28--17-40-52--19

Here’s a route of the errors. The last 5 minutes of the drive is when I start it. 

It involves holding the distance adjustment button and then engaging OP.
(Why I did this I’m not sure lol)

sshane, 2022-06-11:
Should be fixed by https://github.com/commaai/openpilot/pull/24674, let me know if it isn't

You initially triggered this by holding the gap dist button with the accel button, then let go the gap button and the signal transitioned to accel and back to dist gap without reaching zero, so CarState never sent an unpressed event for the accel button. Then you happened to never press it for the rest of the drive, so controlsd thought the accel button was being pressed this whole time 

[image]

closed 2022-06-11 by sshane