commaai/openpilot #23269: Use mypyc to compile controlsd and others

geohot, 2021-12-19:
mypy has a built in Python -> C compiler and it seems decent.

I think it's worth trying to see if we get CPU reduction. It also gets better as you add type annotations.

Might help with #23145

adeebshihadeh, 2021-12-22:
I suspect we'll see little CPU reduction with a bunch of long tail issues trying to do this for whole daemons. This is probably better suited for small, self-contained stuff like CANParser.

[Here's](https://github.com/commaai/openpilot/blob/50cce27e34aaed9ee5029031447b36a8fa328851/selfdrive/test/test_onroad.py#L24) a list with all the current CPU usage for each daemon. controlsd, plannerd, and paramsd are the top python processes by CPU usage. The slow parts of plannerd and paramsd, acados and rednose, are already in C. As for controlsd, I recently fixed the random high CPU usage, and I doubt the high baseline CPU usage has anything to do with controlsd being written in python.

geohot, 2021-12-26:
Why do you think the high sustained CPU usage in controlsd is not due to Python? I imagine clean C++ being 5x more efficient, and bet there's 2x from a good compiler.

adeebshihadeh, 2022-01-02:
Looked into controlsd a bit, and we seem to waste a ton on of CPU on dealing with events and building capnp messages. I think cleaning up the events stuff, fixing the capnp bad practices, and building optimized versions of Python and capnproto will finally make controlsd CPU usage reasonable. I got a 10% improvement in controlsd running a Python built with optimizations (with the default profile tasks). We’re throwing away a good bit of performance by not optimizing these things we use everywhere. 

Yes, that’s still bz2. Definitely need to look into our OMX encoder more. @deanlee want to take a look into it?

As for camerad, I think a bunch of other daemons also use a bunch more CPU on the C3, or at least that was the case earlier on in the bringup. I don’t think we ever looked into any of them though.

deanlee, 2022-01-03:
It looks like that we can get rid of pycapnp after capnp 0.8.0
we can run `capnp compile -ocython log.capnp` to wrap the c++ classes in gen/log.capnp.h with cython, it should be much faster than dynamicstruct based pycapp.

adeebshihadeh, 2022-01-03:
> It looks like that we can get rid of pycapnp after capnp 0.8.0 we can run `capnp compile -ocython log.capnp` to wrap the c++ classes in gen/log.capnp.h with cython, it should be much faster than dynamicstruct based pycapp.
> 
> the cython file generated after run capnp tool: [log_capnp_cython.pyx.tar.gz](https://github.com/commaai/openpilot/files/7803123/log_capnp_cython.pyx.tar.gz)

We have 0.8.0 on AGNOS and NEOS, so we can already ship this. Want to see how much we can speedup controlsd or plannerd with this?

deanlee, 2022-01-04:
I am looking to see if there is way to import the new interface and does not need to touch existing code.

deanlee, 2022-01-20:
`I420Scale` is the most CPU consuming operation in the encoder.  It takes about 5-9ms. (the `I420ToNV12` only takes 0.3-0.5ms). https://github.com/commaai/openpilot/pull/23589  can reduce some cpu usage by using omx to scale the frame.

the IO operations are also very CPU-intensive in encoder.  It can be improved by pre-allocating disk space using fallocate and write to it without buffering.

pd0wm, 2022-02-10:
Think we can also move the writing into a separate thread. We currently block the main encoder thread on IO.

geohot, 2022-02-12:
I wonder if that's responsible for the glitched videos we see sometimes, I don't think we know if the delay is the encoder itself or just the kernel blocking on write. Propose adding a msgq between the encoder thread and the writer thread, with some metadata to command writer thread to rotate the files. This also gives free streamerd with bridge.

adeebshihadeh, 2022-06-02:
I think we can close this now that we have more specific issues open.

closed 2022-06-02 by adeebshihadeh