commaai/openpilot #23278: Nav UI route stats/turn instruction misbehavior

jyoung8607, 2021-12-20:
### Describe the bug

I used openpilot navigation on a drive from Cincinnati to St. Louis. It worked fine in the beginning, but some time into the drive, the nav UI began some manner of freakout. I don't know what triggered it. I was engaged and driving normally, and I don't think I was interacting with the UI when it started.

[image]

The above GIF shows it rapidly cycling between different versions of the travel time, estimated time of arrival, distance remaining, and next turn instruction. I think it may have been cycling through future navigation plan segments. The references to I-70 toward St. Louis were along my planned route, but an hour or more in the future.

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

3cfdec54aa035f3f|2021-12-18--05-01-39

### openpilot version

master @ e4de82f90636a551a2490ccda1abc30a9b7f30b1

### Additional info

The misbehavior, and my attempts to reset and troubleshoot, are in the 05:15-05:30 range of the above route.

I was able to cancel navigation just fine, but on restarting navigation to the same place in St. Louis, the UI misbehaved immediately in the same way. I then tried navigation to two other destinations, one of which was a direct U-turn back home, and the same misbehavior continued. IIRC it was still showing basically the same stats/turn info in rotation, which means it was still stuck from the prior nav destination. I wasn't all that far from home yet, and I-70 wouldn't have been involved in a return route.

pd0wm, 2022-02-15:
This bug turned out to be quite tricky to reproduce. Turns out that for that specific stretch of route the mapbox API doesn't always return the same list of coordinates. I assume due to a bug somewhere on their end or in the Qt parsing code of the polyline. When the response is different one of the coordinates in the path is replaced by the previous one, which in turn breaks the `distance_along_geometry` function, which is used to transition to the next maneuver and showing the distance.

This function internally uses `minimum_distance(a, b, p)` which computes the shortest distance from p to a line segment `a -> b`. If a and b are the same point the result is wrong and always returns 0. I'll add a check for this specific case, which should also fix this bug.

closed 2022-02-15 by pd0wm

jyoung8607, 2022-02-15:
Nice work, that would have taken me an eternity to track down.

Should you ever need a second repro case, `3cfdec54aa035f3f|2022-02-12--23-02-29` somewhere in the segment 28-29-30 area does it again on an entirely different road. It was on my to-do list to write up and add here. The display freakout self-resolved as I came up on the next turn in the planned route, and that behavior aligns well with your explanation and fix.

pd0wm, 2022-02-15:
Let me know if you see it again on latest master! There might be other bugs that would cause the same behavior.

PR #23769 (2022-02-15): nav: handle route responses with duplicate points
Resolves #23278