commaai/openpilot #23281: Navigation map opens when device boots even though no directions were sent

gregjhogan, 2021-12-21:
### Describe the bug

On master, if my device boots when I turn my car on, when openpilot goes on-road the nav map pops open with no directions active (I didn't send any directions).  This seems to consistently happen.

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

5decac85ab437ed1|2021-12-15--07-51-16

### openpilot version

16dd7ce1fe2a2b0537fefd5ad131d1494d0d09ad

jyoung8607, 2021-12-21:
I've been noticing this lately on master as well. I assumed it was an intentional feature and not a bug, perhaps not...?

pd0wm, 2021-12-21:
Introduced by https://github.com/commaai/openpilot/pull/22789

closed 2021-12-21 by adeebshihadeh