commaai/openpilot #23282: Nav/Athena behavior when disconnected

jyoung8607, 2021-12-21:
### Describe the bug

Tried to use navigation on the return leg of my road trip Sunday and had several weird issues:

1. The UI showed connect offline, although I seemed to be in good LTE coverage
2. I chose my preprogrammed home location from the C3 UI as my nav destination
3. Navigation never really started, but the Nav UI pane did come up
4. I tried to cancel navigation, but the Nav menu disappeared in favor of the Nav/Prime ad screen
5. Rotating among UI views, the Nav UI pane was still accessible despite this
6. If I selected the usual full-screen onroad view, the Nav pane would reappear on its own within about a minute

I drove in this state for about 90 minutes. To get connectivity back and make the nav pane stop popping up, I had to reboot the device at my next stop.

There were a bunch of times I tried to dismiss the Nav pane and had it come back. The instances I made note of were right at the end of the route, about 6:36PM and 6:41PM, it popped back up after about a minute each time. I actually had the UI process die and restart on me right at the end too, but that could be the usual watchdog thing.

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

`3cfdec54aa035f3f|2021-12-19--17-16-31`

### openpilot version

master @ 16dd7ce1fe2a2b0537fefd5ad131d1494d0d09ad

### Additional info

There's several issues to unpack here.

The C3 UI showed I had good LTE coverage. I don't know if I had working Internet connectivity; I wasn't in a position to SSH in and check. Athena certainly thought I was offline. Device reboot was definitely needed to get it back.

Earlier that day, several drive-cycles (but not power-cycles) prior, I did spend time in an area that legitimately had no cell service. I drove into the area, stopped and parked for an hour, and started a new drive out of that area. I don't know when the Athena no-connectivity issue started, but it's possible that it never recovered from earlier that day.

Even if I had no Internet connectivity, the UI shouldn't have treated me as if I weren't a Prime/Nav subscriber.

It shouldn't be possible to lose the Nav settings/destination menu while still having the Nav onroad pane in rotation.

The Nav pane shouldn't have kept popping up unsolicited. That was weird. Maybe a nexus with #23281?

michaelhonan, 2021-12-21:
This happens to me with my comma two. Unsure if it's the same issue, but after a day (usually when I get back into the car of a morning), it'll say the device is offline. Requires a reboot to get connection back.

LTE reports full signal during this time.

tabdallah, 2022-01-29:
I have noticed this same behavior several times this week while on a cross-country drive. If I start a drive cycle in an area with no cell coverage, the device requires a hard power cycle to restore connect services once cell coverage is restored. The C3 indicates cell coverage correctly, but connect remains offline across multiple drive cycles until I perform a hard reset.

PR #23770 (2022-02-15): nav: set settings stack based on uiState.prime_type
resolves #23282

page might still be empty if there is no cached api response, but it won't go to the add after the first response times out.
closed 2022-02-15 by pd0wm