commaai/openpilot #23284: paramsd: bad init while in a turn

adeebshihadeh, 2021-12-21:
Device booted while in a turn and got vehicle parameters identification failed for about the first minute of bd6a637565e91581|2021-12-20--17-43-09

sshane, 2021-12-22:
Same here, I restarted the car while driving to flip a toggle and had the vehicle parameter identification failed as well, but it's not a stock route.

nuwandavek, 2022-01-05:
@adeebshihadeh Do you have the rlogs for this route?

The cause for almost all of these alerts now is : filter becomes "active" in a turn (the yaw rate observation is thus very very high) -> the filter state takes some small time to go from it's earlier value (0, likely) to this new high value. In the meantime, the offset is the only free variable that can move rapidly, and picks up the slack to keep the vehicle model consistent.
The roll-comp reduces this effect a bit, but does not eliminate it. Can you upload the rlogs?

pd0wm, 2022-01-17:
@nuwandavek Fixed by https://github.com/commaai/openpilot/pull/23449?

nuwandavek, 2022-01-17:
No, this is unrelated. Roll comp should reduce the occurrence of this issue, but does not eliminate it. We should rethink the states vs observations in the filter. We'll work on this next!

nuwandavek, 2022-03-17:
This PR https://github.com/commaai/openpilot/pull/23951 should fix this issue. The cause, in most cases was due to a delay in `cameraOdometry` messages. When the filter was initialized in turns, without pose information, the gyro bias took on a very high value, which took ~1-2 mins to converge. This caused wrong estimation of YR, resulting in incorrect fast-angle-offset values.

closed 2022-03-17 by nuwandavek