commaai/openpilot #23286: 0.8.12 takes exits

alancyau, 2021-12-21:
### Describe the bug

I experienced the car taking an exit twice. It appears that the lead car taking an exit will make the OP model want to follow it.

a4af1602d8e668ac|2021-12-21--16-29-00--12
a4af1602d8e668ac|2021-12-21--16-29-00--16

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

a4af1602d8e668ac|2021-12-21--16-29-00--12

### openpilot version

0.8.12

tecandrew, 2022-01-04:
experienced the same thing yesterday. 

route: `eb5a50f9e98f50b9|2022-01-03--15-23-00--12` on comma three, laneless

https://user-images.githubusercontent.com/21319730/148128976-0fddfe8b-8548-43c3-9db0-3ebb1791ffc5.mov

https://user-images.githubusercontent.com/21319730/148128979-d7ffdaf9-3cda-44eb-b772-dbe3aabfc477.mov

adeebshihadeh, 2022-01-04:
@mitchellgoffpc feel free to move this to a discussion if this is more of a long term project

adeebshihadeh, 2022-06-02:
Both 0.8.13 and 0.8.14 have new models that shouldn't take exits. Feel free to report back if you're still having issues.

closed 2022-06-02 by adeebshihadeh