commaai/openpilot #23298: plotjuggler crashes when opening lat/long tuning layout

gregjhogan, 2021-12-22:
### Describe the bug

```
$ cd ~/openpilot
$ ./update_requirements.sh 
# output omitted
$ scons -j$(nproc)
$ cd ~/openpilot/tools/plotjuggler
$ ./install.sh
# output omitted
$ PYTHONPATH=~/openpilot ./juggle.py --demo --layout ./long-tuning.xml 
Loading https://commadata2.blob.core.windows.net/preserve/4cf7a6ad03080c90/2021-09-29--13-46-36/0/rlog.bz2?se=2021-12-22T23%3A37%3A33Z&sp=r&sv=2018-03-28&sr=b&rscd=attachment%3B%20filename%3D4cf7a6ad03080c90_2021-09-29--13-46-36--0--rlog.bz2&sig=C1OCwMaDVGukE%2B2gX0pL8i/OfhdUCDFrZ5O5vw%2BPvjM%3D
Loading https://commadata2.blob.core.windows.net/preserve/4cf7a6ad03080c90/2021-09-29--13-46-36/1/rlog.bz2?se=2021-12-22T23%3A37%3A33Z&sp=r&sv=2018-03-28&sr=b&rscd=attachment%3B%20filename%3D4cf7a6ad03080c90_2021-09-29--13-46-36--1--rlog.bz2&sig=55%2BUqwsJJZFF1hk%2BE2SJWDikn7idX4fcKGe9uJO0pIU%3D
Loading https://commadata2.blob.core.windows.net/preserve/4cf7a6ad03080c90/2021-09-29--13-46-36/2/rlog.bz2?se=2021-12-22T23%3A37%3A33Z&sp=r&sv=2018-03-28&sr=b&rscd=attachment%3B%20filename%3D4cf7a6ad03080c90_2021-09-29--13-46-36--2--rlog.bz2&sig=h3rmTb5qi6vguBaYXQyO9s419pyR%2By3p5IbzSAmZfUU%3D
Loading https://commadata2.blob.core.windows.net/preserve/4cf7a6ad03080c90/2021-09-29--13-46-36/3/rlog.bz2?se=2021-12-22T23%3A37%3A33Z&sp=r&sv=2018-03-28&sr=b&rscd=attachment%3B%20filename%3D4cf7a6ad03080c90_2021-09-29--13-46-36--3--rlog.bz2&sig=ggk0dWKUzn6GvSAO%2B0dO2cwvD1aU825QadI%2B/hWK6ag%3D
Loading https://commadata2.blob.core.windows.net/preserve/4cf7a6ad03080c90/2021-09-29--13-46-36/4/rlog.bz2?se=2021-12-22T23%3A37%3A33Z&sp=r&sv=2018-03-28&sr=b&rscd=attachment%3B%20filename%3D4cf7a6ad03080c90_2021-09-29--13-46-36--4--rlog.bz2&sig=IXu0fIPXFfISiwQdwjTlq5jEKL99BDmZ1AEvWXsn5VY%3D
Loading https://commadata2.blob.core.windows.net/preserve/4cf7a6ad03080c90/2021-09-29--13-46-36/5/rlog.bz2?se=2021-12-22T23%3A37%3A33Z&sp=r&sv=2018-03-28&sr=b&rscd=attachment%3B%20filename%3D4cf7a6ad03080c90_2021-09-29--13-46-36--5--rlog.bz2&sig=RQMG3MhfBsNwdtCEG1DD1iWjxUwYhatjBWH7mrARBDk%3D
Loading https://commadata2.blob.core.windows.net/preserve/4cf7a6ad03080c90/2021-09-29--13-46-36/6/rlog.bz2?se=2021-12-22T23%3A37%3A33Z&sp=r&sv=2018-03-28&sr=b&rscd=attachment%3B%20filename%3D4cf7a6ad03080c90_2021-09-29--13-46-36--6--rlog.bz2&sig=0J/gBd8VxG1mCKyBCz%2BwVGaBWsjW%2BUVJm/DUDz1PqTk%3D
Loading https://commadata2.blob.core.windows.net/preserve/4cf7a6ad03080c90/2021-09-29--13-46-36/7/rlog.bz2?se=2021-12-22T23%3A37%3A33Z&sp=r&sv=2018-03-28&sr=b&rscd=attachment%3B%20filename%3D4cf7a6ad03080c90_2021-09-29--13-46-36--7--rlog.bz2&sig=l30ok0HLN4FtiP%2BMRjiCIFK7pAhIorf8ViFWA2lgftY%3D
Loading https://commadata2.blob.core.windows.net/preserve/4cf7a6ad03080c90/2021-09-29--13-46-36/9/rlog.bz2?se=2021-12-22T23%3A37%3A33Z&sp=r&sv=2018-03-28&sr=b&rscd=attachment%3B%20filename%3D4cf7a6ad03080c90_2021-09-29--13-46-36--9--rlog.bz2&sig=e0CzV6Kje49n0gNiDynqU4UMVf7UhwCaq8L%2BcWDIrwo%3D
Loading https://commadata2.blob.core.windows.net/preserve/4cf7a6ad03080c90/2021-09-29--13-46-36/10/rlog.bz2?se=2021-12-22T23%3A37%3A33Z&sp=r&sv=2018-03-28&sr=b&rscd=attachment%3B%20filename%3D4cf7a6ad03080c90_2021-09-29--13-46-36--10--rlog.bz2&sig=Y1VqH7FiWMzeeaxTIeLg9diH7a2Se/sP0i02oDMzBhM%3D
Loading https://commadata2.blob.core.windows.net/preserve/4cf7a6ad03080c90/2021-09-29--13-46-36/8/rlog.bz2?se=2021-12-22T23%3A37%3A33Z&sp=r&sv=2018-03-28&sr=b&rscd=attachment%3B%20filename%3D4cf7a6ad03080c90_2021-09-29--13-46-36--8--rlog.bz2&sig=uLXUxQ50Z0Llnc9CpVK29JoHQ%2BlWtz6hmL/NlFC3rWo%3D
100%|████████████████████████████████████████████████████████| 491850/491850 [00:01<00:00, 334116.96it/s]
Loading compatible plugins from directory:  "/home/greg/openpilot/tools/plotjuggler/bin"
"libDataLoadRlog.so is a DataLoader plugin"
Loaded DBC: toyota_rav4_2017_pt_generated
Loading schema: "/home/greg/openpilot/cereal/log.capnp"
"libDataStreamCereal.so is a DataStreamer plugin"
Number of plugins loaded:  2 

Loading compatible plugins from directory:  "/home/greg/.local/share/PlotJuggler"
Number of plugins loaded:  0 

Loaded DBC: toyota_rav4_2017_pt_generated
Loading schema: "/home/greg/openpilot/cereal/log.capnp"
Done reading Rlog data
Stack trace (most recent call last):
#9    Object "", at 0xffffffffffffffff, in 
#8    Object "/home/greg/openpilot/tools/plotjuggler/bin/plotjuggler", at 0x55a30d6c54bd, in _start
#7    Source "../csu/libc-start.c", line 308, in __libc_start_main [0x7fdea1d6f0b2]
#6    Source "/tmp/plotjuggler/plotjuggler_app/main.cpp", line 322, in main [0x55a30d6c4046]
#5    Source "/tmp/plotjuggler/plotjuggler_app/mainwindow.cpp", line 267, in MainWindow [0x55a30d6ff66a]
#4    Source "/tmp/plotjuggler/plotjuggler_app/mainwindow.cpp", line 1972, in loadLayoutFromFile [0x55a30d6fcc93]
#3    Source "/tmp/plotjuggler/plotjuggler_app/transforms/lua_custom_function.cpp", line 126, in xmlLoadState [0x55a30d753d93]
#2  | Source "/tmp/plotjuggler/plotjuggler_app/transforms/lua_custom_function.cpp", line 26, in operator=
    | Source "/tmp/plotjuggler/./3rdparty/sol/sol.hpp", line 19191, in operator=
    | Source "/tmp/plotjuggler/./3rdparty/sol/sol.hpp", line 15325, in operator=
    | Source "/tmp/plotjuggler/./3rdparty/sol/sol.hpp", line 15266, in operator=
    | Source "/tmp/plotjuggler/./3rdparty/sol/sol.hpp", line 9444, in move_assign<false>
    | Source "/tmp/plotjuggler/./3rdparty/sol/sol.hpp", line 9297, in deref
    | Source "/tmp/plotjuggler/./3rdparty/sol/sol.hpp", line 9333, in deref
      Source "/tmp/plotjuggler/./3rdparty/sol/sol.hpp", line 9259, in initEngine [0x55a30d75464f]
#1    Source "/tmp/plotjuggler/3rdparty/lua-5.4.1/src/lauxlib.c", line 670, in luaL_unref [0x55a30d7c3776]
#0  | Source "/tmp/plotjuggler/3rdparty/lua-5.4.1/src/lapi.c", line 714, in finishrawget
      Source "/tmp/plotjuggler/3rdparty/lua-5.4.1/src/lapi.c", line 682, in lua_rawgeti [0x55a30d7adfed]
Segmentation fault (Address not mapped to object [0x8])
Segmentation fault (core dumped)
```

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

any route (demo route used above)

### openpilot version

master

adeebshihadeh, 2022-01-01:
This seems to be an issue with PlotJuggler, since opening the layout after `./juggle.py --demo` works fine.

gregjhogan, 2022-01-01:
Maybe we just need to regenerate the saved layout for the latest version?

gregjhogan, 2022-01-01:
Did you go through the install instructions and make sure everything is up to date? If I fire it up then open the layout, it crashes in the same way when I open the layout.

adeebshihadeh, 2022-01-01:
yeah, I rebased our fork today. Did you update recently?

gregjhogan, 2022-01-01:
When it was crashing I know it didn't matter, but I had not updated/reinstalled in the last few hours, and after doing so I see it seems much better now and I can open the layout after opening plotjuggler (without specifying `--layout`)!

Now now the only issue is if I do specify `--layout` I get this dialog:
[image]

adeebshihadeh, 2022-01-01:
Ah yeah, I was seeing that too. No clue what it means.

gregjhogan, 2022-01-01:
It was a bug we must have introduced at some point and I fixed it in #23348

But after fixing that I still get this messages, maybe the layout needs to be re-generated for the latest schema or something.
[image]

adeebshihadeh, 2022-01-03:
Crashing seems fixed, and the curve missing isn't really an issue. It's complaining that `controlsState.lateralControlState.pidState.saturated` is missing because LQR is used instead of PID on the demo route.

closed 2022-01-03 by adeebshihadeh