commaai/openpilot #23302: openpilot stops too far back from lead on incline

gregjhogan, 2021-12-23:
openpilot seems to consistently stop further back on an incline compared to flat ground
example: `5decac85ab437ed1\|2021-12-22--12-29-05--16`
first engagement it stops too far back, then next 3 I try to get it to creep forward
[image]
once stopped too far back, I see it oscillates between the stopping (longControlState==2) and starting (longControlState==3) state @ 50 Hz
[image]

sshane, 2021-12-24:
I believe the stopping/starting oscillation is due to `vEgoStopping` being greater than `vEgoStarting`, which will always lead to a state transition oscillation due to the logic in longcontrol.py (assuming this car is a Hyundai, the route isn't public to check out).

pd0wm, 2021-12-24:
Think I added the high vEgoStopping because the acceleration started oscillating a lot around the set point at low speeds. But maybe we should check if that's still the case, since there probably have been a bunch of changes since then.

closed 2022-11-16 by HaraldSchafer